prusa3d/PrusaSlicer #12972: esta dando pra ve estrura como antigamente na area de fatiamento como aquela barra de rodagem

andrelimapoeta, 2024-06-30:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

andrelimapoeta, 2024-06-30:
esta ruim essa versao 
barra de fatiamento ruim 
nao da ve fatiamento dos objetos

malowz, 2024-06-30:
**Português:** você quer dizer que após fatiar um objeto, nao aparece nada na mesa? é um problema que já está sendo discutido aqui. temos que esperar eles verificarem o erro e corrigir numa proxima versão

**English** He is most likely complaining about the bug that does not show the sliced part on the bed.

closed 2026-09-29 by prusaslicer-bot[bot]