prusa3d/PrusaSlicer #12973: Several Problems launching 2.8.0 AppImage as upgrade from 2.7.x

TheWebMachine, 2024-06-30:
### Description of the bug

Coming from v2.7.4, then trying to run the new AppImage for 2.8.0, it would simply not load. Attempting to run in terminal showed two causes, only one of which I could correct:

````
./PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929_dfd3cdac76b94d77e7d6b2eac3000f92.AppImage 
[2024-06-30 02:31:04.604465] [0x00007f3b6fa57c40] [error]   Error in a Vendor Config Bundle "/home/t***e/.config/PrusaSlicer/vendor/PrusaResearch.ini": The printer preset "printer:Original Prusa SL1S SPEED" contains the following incorrect keys: area_fill, which were removed
[2024-06-30 02:31:04.734580] [0x00007f3b6fa57c40] [error]   Error in a preset file: The preset "/home/t***e/.config/PrusaSlicer/printer/Anycubic Photon D2.ini" contains the following incorrect keys: area_fill, which were removed
[2024-06-30 02:41:00.409084] [0x00007f2b2cf3fc40] [error]   UserAccount: Failed to read token - no datafile found.
terminate called without an active exception
````
I was able to comment out the `area_fill` fields where indicated and those errors are gone. Shouldn't the new version automatically remove this depreciated value from the profiles on first launch? 

However, there isn't much I can do about the `UserAccount` error. Since a token can't be created until the app is launched and a user signs in, shouldn't this also be set to a null/ignored value on first launch of the new version?

### Project file & How to reproduce

"Upgrade" to the new 2.8.0 AppImage
Launch 2.8.0
Will fail to launch because of a missing UserAccount token, but no error is shown to user unless ran from terminal

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929

### Operating system

Ubuntu 20.04.6-LTS

### Printer model

Prusa Mini+, Creality Ender 3 S1 Plus, Creality K1C, Anycubic Photon D2

lukasmatena, 2024-06-30:
Thanks for the report. Can you please run it with `--loglevel 5`?

Icelk, 2024-06-30:
The UserAccount thing doesn't seem to be connected. I also get it using 2.8 but the application starts just fine. Maybe try restarting your computer, which fixed it for me (though I'm running on Arch with custom kernel modules (dkms for nvidia drivers), so maybe it's just your graphics drivers not functioning?)

closed 2026-09-29 by prusaslicer-bot[bot]