prusa3d/PrusaSlicer #12975: PrusaSlicer show "Failed to load a web browser" in physical printer menu

Fsigr, 2024-06-30:
### Description of the bug

PrusaSlicer show "Failed to load a web browser" in physical printer menu. In "Printer" menu all ok.
[image]
I use Win10 x64 and default browser Vivaldi latest version

### Project file & How to reproduce

1. Use Win10 x64 and default browser Vivaldi latest version
2. Add manualy 3D printer Ender3 V3 KE in menu "Printers", tested it ok
3. Go to menu "Physical printer" and see error

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.8.0+win64

### Operating system

Windows 10 x64

### Printer model

Creality Ender3 V3 KE

dwprod38, 2024-06-30:
New PrusaSlicer 2.8.0 had warning message after launching application " Error running JavaScript: TypeError: undefined is not an object (evaluating 'window.webkit.messageHandlers')" and impossible to view layers overview after slicing generation... (MacOS 10.12.6)

Back to Prusaslicer 2.7.4 and works fine...

 [
[image]
](url)

buhman, 2024-09-03:
related #12968

closed 2026-09-29 by prusaslicer-bot[bot]