prusa3d/PrusaSlicer #12976: V2.8.0 - Log in with Apple does absolutely nothing.

borgsamuel, 2024-06-30:
### Description of the bug

After updating to version 2.8.0, I went to use the new log in toggle in the app. I log in usually using my apple id and it always works but on the slicer it just takes my details, closes the tab and nothing else happens. Relaunching and reinstalling didn't change a thing. 

### Project file & How to reproduce

No project was in use. To reproduce the error, just log in with your apple id instead of the other log in choices.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.0+MacOS-x64

### Operating system

macOS Sonoma 14.5

### Printer model

Prusa MK4

fergymcferguson, 2024-07-02:
+1 here - was working in the beta, but now it just sits on the login screen after using apple login option.

fergymcferguson, 2024-07-17:
[image]
Still sitting here waiting for it to log in since this issue was originally posted. 👎🏼

StephanPernkopf, 2024-09-08:
I get the same issue on my mac with apple silicon (`PrusaSlicer-2.8.0+MacOS-arm64-202406270929`)

I wanted to look at the logs of prusaslicer so I started it like this `./PrusaSlicer --loglevel=9`
And I get this error log line right as I click on **Continue with Apple**:
`[2024-09-08 20:15:50.764312] [0x00000001ec3be940] [error]   WebViewDialog error: wxWEBVIEW_NAV_ERR_OTHER`

yaronalk, 2024-09-30:
Same, please fix it

StephanPernkopf, 2025-01-05:
The latest version (`2.9.0`) seems to have fixed the issues for me. 
I can login now using apple id.

closed 2026-09-29 by prusaslicer-bot[bot]