prusa3d/PrusaSlicer #12977: v2.8.0 GTK3 Icons much Smaller (too small)

LightningStalker, 2024-06-30:
**Is your feature request related to a problem? Please describe.**
Icons are small on 4k/8k screens (even max custom size) from GTK3 AppImage

**Describe the solution you'd like**
Adjustment range >100% custom icon size

**Describe how it would work**
Menu, Configuration, Preferences, GUI, use custom size for toolbar icons and adjustment slider greater size

**Describe alternatives you've considered**
Compile patch from source

**Additional context**
Download 2.8.0

psi-4ward, 2024-11-13:
[image]

I'm on Arch-Linux using KDE+Wayland on 4k screens. Would be very nice to have an option to increase the icon size.

LightningStalker, 2024-11-14:
I'm noticing a lot of programs now a have UI scaling that goes >100%.

LightningStalker, 2026-09-03:
The main problem I'm haveing now is no way to close the Settings windows with Ctrl+q, and no button to close them.