prusa3d/PrusaSlicer #12978: Infill direction corresponding to layers option suggestion

BoredCoat, 2024-06-30:
What I have tried to do and support told me it is impossible to do atm is to **change infill and angle of it every X layers**. Like in this picture, I would like to have two layers and the another two alternating like this (this is a picture how **IT DOESNT WORK**, I would like it to change every two, not three):
[image]

**Solutions:**
Make a menu to customize those type of parameters or make height range modifier work with this (it does not at the moment)

**How it would work:**
I imagine a menu that lets you set a rotation angle and type of infill for a defined layer range (ex. for layer height of 0.2: 0-0.4 mm; 0°; alligned rectalinear,  0.4-0.8; 90°; alligned rectalinear)

**Alternatives:**
Make height range modifier work. If I set 0-0.4 (2 layers) to have alligned rectalinear, why does it make 3 layers alligned (like it the picture) and not the amount I set.

u89djt, 2024-07-01:
(fellow user passing through) Post the 3mf file (zipped) that corresponds to your screenshot so that whoever looks at this can see what you've tried.

neophyl, 2024-07-01:
You could also try the combine infill setting.  If you combined 2 layers of infill the angle would change every 2 layers.  At least it would alternate every x layers.  Doesn't help with the angle part of it and it wouldn't will aligned rectilinear as that is designed not to change angle at all, which is the whole point of the 'aligned' part  but combining normal rectilinear might give you what you want.

closed 2026-09-29 by prusaslicer-bot[bot]