prusa3d/PrusaSlicer #12980: .tar of new tags prusaSlicer

otaolafranc, 2024-06-30:
### Description of the bug

Hello,
since 2.8.0 rc2 the tar version on the tag disappear, I would like that this format comes back as it allows updating my system quite easily when it comes to prusaSlicer.

### Project file & How to reproduce

the new assets dont have this format since 2.8.0-rc1

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

linux Tuxedo OS 2

### Printer model

mk4

VuiMuich, 2024-07-01:
Also packaging for NixOS would be much easier from a tarball, then from the AppImage.

otaolafranc, 2024-07-27:
Is this going to be corrected for next release?

VuiMuich, 2024-07-28:
It appears, that for NixOS the package is actually built from a cloned source and not as assumed from the tarball.

closed 2026-09-02 by otaolafranc