prusa3d/PrusaSlicer #12981: Have seams visible in preview by default or let us toggle it on or off!

JagguRaja, 2024-06-30:
**Is your feature request related to a problem? Please describe.**
No problem, just an annoyance. I hate having started a print only to realize the seam is in a terrible location. Sometimes, in a rush I don't see the seams in the preview.

**Describe the solution you'd like**
Have seams visible in preview by default or let us toggle it on or off.

**Describe how it would work**
Toggle button. I don't care how it's implemented. Or just on my default as the seam is important to many.

neophyl, 2024-06-30:
Please search for existing issues before opening new ones.  This is a duplicate.  For example
https://github.com/prusa3d/PrusaSlicer/issues/9468
or how about https://github.com/prusa3d/PrusaSlicer/issues/10344
or https://github.com/prusa3d/PrusaSlicer/issues/11952
or an even earlier one https://github.com/prusa3d/PrusaSlicer/issues/8418

JagguRaja, 2024-06-30:
> Please search for existing issues before opening new ones. This is a duplicate. For example #9468 or how about #10344 or #11952 or an even earlier one #8418

Nothing was done about them so figured adding one again might renew the conversation. We been asking for the feature for years.

mix579, 2024-07-01:
This feels like a simple thing to implement and has been requested by many people. I wish the slicer team would dedicate at least some of their efforts to small but meaningful improvements like this one.

JagguRaja, 2024-07-01:
Exactly! Like who requested a reskin? Improvements like these are what we
need!

mix579, 2024-07-01:
Yeah…. I didn't mention it because I for one actually like the reskin (in particular the Preview window) but frankly, I'd rather they FIRST focused on suggestions like the visible seam , straight line seam painting, etc., than stuff no one (as far as I can tell) had asked for.

mildmojo, 2024-07-03:
Are you looking for the seam view toggle in the slicing preview legend window? It's the button that looks like a broken chain link, and it makes the seams show up in bright white. (I've toggled it on for this screenshot.) 

[image]

It stays on until I close the app (v2.8.0), even if I re-slice the model or load a different project. I'm not sure if you're looking for the feature itself, or a way to make it turn on automatically each time you start the app (like the other tickets neophyl linked).

JagguRaja, 2024-07-03:
Thanks but I'm not looking for the feature itself. I'm looking for a way to
have it on by default. Like you said, once you close the app and open it up
again, you need to re-enable the seam to see it.

mildmojo, 2024-07-03:
Cool. I tend to leave the app running for weeks at a time across multiple projects, so I hadn't thought of saving view settings. Sounds like a good enhancement.

kkhalme, 2024-09-29:
The basic workflow in PrusaSlicer could use a lot of enhancements, including the configuration of default legend options (not just seams) and also keyboard shortcuts for quickly toggling specific legends, just to mention a few.

closed 2026-09-29 by prusaslicer-bot[bot]