prusa3d/PrusaSlicer #12982: Empty Preview in Prusa Slicer 2.8.0

dwprod38, 2024-06-30:
### Description of the bug

New PrusaSlicer 2.8.0 had warning message after launching application " Error running JavaScript: TypeError: undefined is not an object (evaluating 'window.webkit.messageHandlers')" and impossible to view layers in "Preview" after slicing generation... (MacOS 10.12.6)

Back to Prusaslicer 2.7.4 and works fine...

[image]

[image]
[image]

### Project file & How to reproduce

1- Place Stl object
2- Click on "Slice Now" button ... generation finished
3- Click preview button : none object with layers is displayed on bed but generated gcode file is present...

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer Version 2.8.0+MacOS-x64

### Operating system

MacOS 10.12.6

### Printer model

MK4 + MMU3

neophyl, 2024-06-30:
Duplicate issue.  https://github.com/prusa3d/PrusaSlicer/issues/12908

lukasmatena, 2024-07-04:
Closing as a duplicate of #12908.

closed 2024-07-04 by lukasmatena