prusa3d/PrusaSlicer #12983: Layer lines at same position every print.

rinkek, 2024-06-30:
### Description of the bug

Every print there are some odd lines at layer start. This seems some kind of pressure advance issue. I don't have this on 2.7.4, which is still have on my laptop.

Printed with the -exact- same settings, the two on the ledt printed with 280, the two on the right printed with 274 and OrcaSlicer.

[image]

### Project file & How to reproduce

I just printed this xyz-cube after upgrading to 280, prints fine with 274, same settings.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

280

### Operating system

Windows 11

### Printer model

Voron Trident 250

u89djt, 2024-07-01:
(fellow user just passing through) post the project files in 2.8.0 and 2.7.4 so whoever looks at this knows what your various parameters are. I sort of wonder if seams might be part of your issue? Very amateur speculation.

rinkek, 2024-07-01:
I think i already had an issue and 280 is showing this issue better than 274.

I think this is not related to 280

rinkek, 2024-07-01:
I think this is not related to 280

rinkek, 2024-07-02:
It was related to Z-HOP. When turned off the print was much better.
However, with 2.7.4 it completely gone, with 2.8.0 it's still visible, albeit very small.
I think 2.8.0 needs some improvement on seams.

neclimdul, 2024-07-06:
@rinkek which seam alignment are you using?

rinkek, 2024-07-07:
> @rinkek which seam alignment are you using?

I'm using aligned, arachne and seam painting on left back corner.

neclimdul, 2024-07-08:
ok that's useful. I was seeing similar issues and I thought it might be the changes made to the placement for rear seams which seems to now avoid corners instead of preferring them. But if you're painting the changes to alignment wouldn't matter.

Talking in the discord, some said the placement was changed as well as "hierarchy of printer moves involving layers and seams" and it sounds like that's a more likely candidate for the cause now.

closed 2026-09-29 by prusaslicer-bot[bot]