prusa3d/PrusaSlicer #12984: 2.8 AppImage Missing libwebkit2gtk-4.0

mkasberg, 2024-07-01:
### Description of the bug

On Ubuntu 24.04 LTS, `PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929.AppImage` fails to run with the following error:

> /tmp/.mount_PrusaSUVwCnm/usr/bin/bin/prusa-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

Notably, `PrusaSlicer-2.7.4+linux-x64-GTK3-202404050928.AppImage` runs fine on the same machine.

I found [this PR](https://github.com/prusa3d/PrusaSlicer/pull/12869) that mentions adding it as a dependency, but (per those docs) I'd have expected it to be a dev dependency for building from source, not a dependency of the published AppImage.

### Project file & How to reproduce

1. Download `PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929.AppImage`
2. `chmod +x PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929.AppImage`
3. `./PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929.AppImage`

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0+linux-x64-GTK3-202406270929

### Operating system

Ubuntu 24.04

### Printer model

Prusa MINI

matbee-eth, 2024-07-02:
the whole release is fucked

mw46d, 2024-07-03:
Did you try to use https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.8.0/PrusaSlicer-2.8.0+linux-x64-GTK3-202406271021-Ubuntu-24-04.AppImage ? And at least for my 22.04, I had to 'apt install' the libwebkit2gtk!
The text at https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0 is somewhat misleading, I believe. They talk about 24.04 with  newer webkit, but then erroneously switch to using 20.04,  I believe.
I hope, that helps a bit;-)

mkasberg, 2024-07-04:
Yes, this is a duplicate of #12931. The fix there worked, using Ubuntu-24-04.AppImage.

closed 2024-07-04 by mkasberg