prusa3d/PrusaSlicer #12988: Can't translate in negative direction anymore

EinsteinFrizz, 2024-07-01:
### Description of the bug

When I try to reposition objects via the Part Manipulation pane it doesn't let me type in negative values (as in, it won't let me input a dash/hyphen) to move them backwards along the axis. I have the same problem with rotation around the axes in the same pane.
[image]
Screenshot of the pane I am talking about - the Position and Rotate (relative) numbers are the ones affected. Obviously I can't screenshot not being able to type in a character but I am using the same US English keyboard layout on Windows 11 that I have used without issue in the past so it is not just the keyboard inputting a weird character instead of the usual dash.

### Project file & How to reproduce

[bugreport-nonegativesallowed.zip](https://github.com/user-attachments/files/16051517/bugreport-nonegativesallowed.zip)
Steps: try to type in a negative number to any of the Part Manipulation Position or Rotation (relative) fields for any of the parts or the object itself.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0+win64

### Operating system

Windows 11 23H2

### Printer model

Anycubic Kossel Linear Plus

u89djt, 2024-07-03:
Same OS and version and I can't type negative numbers into your project either. Here's a workaround while you're waiting for it to be addressed: type your negative number into something else (I just typed it into my browser bar) and you can copy that and paste it into the numerical input and it works.

EinsteinFrizz, 2024-07-06:
Update: I also can't type hyphens in the part names (in a different project)

u89djt, 2024-07-06:
Does the copy/paste somersault work?

lukasmatena, 2024-07-10:
Closing in favor of #13043, which contains a list of all the duplicates.

closed 2024-07-10 by lukasmatena