prusa3d/PrusaSlicer #12990: dans 2.8.0 manque la bande de défilement par couche en dessous du dessin

18pool49, 2024-07-01:
### Description of the bug

la bande de défilement par couche a disparue dans la V : 2.8.0
elle était en dessous du dessin 
il est pas possible de programmer des couches de couleur différente ((RAYURES))  >>> ( en XX couches ou en mm ) ou en aléatoire 
il est prévu les couches en hauteur variable dans les para mais le bouton d'activation est invisible a l'écran 
[image]
[image]
[image]
[image]

### Project file & How to reproduce

[Tornade 5 couleurs projet.zip](https://github.com/user-attachments/files/16053494/Tornade.5.couleurs.projet.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

W11

### Printer model

XL 5

YuSanka, 2024-07-10:
Google translate:

**in 2.8.0 missing the scroll strip by layer below the drawing**

the scrolling strip per layer disappeared in V: 2.8.0
she was below the drawing
it is not possible to program layers of different colors ((STRIPES)) >>> (in XX layers or in mm) or randomly
layers of variable height are provided in the paras but the activation button is invisible on the screen

YuSanka, 2024-07-10:
@18pool49 could you be more specific? What do you actually missed in Slider?

I suppose, that functionality that you are looking for is "Set extruder sequence for the entire print".
Problem is, that this functionality is allowed only for print without any MM painting on it.
If you delete "Multimaterial painting" from the object, than this menu item will appear under cog icon.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]