prusa3d/PrusaSlicer #12995: Organic support interface layers printing in the air

c64clubk, 2024-07-02:
### Description of the bug

Support interface layer is printed not only on/between the tops of branches but also under whole supported area. No matter the count of interface layers (1-3). No matter if generated automatically or painted, no matter  if painted on whole supported area or by many dots. Only few branches existr and then the interface layer is being printed in the air with no pre-support:

[image]

Generating support layer in air makes organic supports unusable for many models, as the model (or another interface layer) also cannot be built on virtual layer underneath.

Interface layer would be much more reliable (or usable at all) if there were more branches under whole supported area, connected with some bridhes where support layer could hang:

[image]

Then form a kind of ladder in first interface layer:
[image]

Then form an interface layer on this ladder instead of in the air:
[image]

### Project file & How to reproduce

Any project with Organic Supports used.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4-linux, 2.8.0-linux

### Operating system

Windows, Linuxmint - different versions

### Printer model

any

u89djt, 2024-07-02:
(interested fellow user) You need to zip the 3mf file to post it...

c64clubk, 2024-07-02:
Practically no need. Just take ANY model with some flat overhang and you get the same result. We have tried on different computers with different models, since Organic supports appeared in precompiled version of PS.

u89djt, 2024-07-02:
Are you using all default parameters?

u89djt, 2024-07-02:
Re "claim": I'm not affiliated with Prusa. I'm here to learn about issues before they affect me when I'm trying to get stuff finished.

u89djt, 2024-07-02:
The project file in holder.zip applies to a MK3S using a 0.6mm nozzle slicing to a layer depth of 0.4mm. Organic support "Branch Density" is set to 10 rather than the more common default of 30.
Left is the slice according to the posted project, right is with branch density set to 30.
[image]

c64clubk, 2024-07-03:
Thanks. Something went wrong with my profile collection. Changed to default 30. 
But even with density 30 (or any other), there is still big amount of small zigzags printed in the air, as tops of branches are not connected before.

u89djt, 2024-07-03:
Yeah, there's a lot of work by smart people gone into this. If you play with the settings and watch what happens, there's a good set of compromises that you can tune for your own printer setup and filament behaviours.
See how most of the extrusions meet a support tip:
[image]
Anything that misses is squished against its neighbour so it can droop, but not entirely. Stretch your object in X and Y and see what it does with a larger area. The main supports are spaced out fairly evenly.
[image]
You're printing at a 0.4mm depth and 0..6mm extruder, so I don't know what maximum branch angle you'd manage, but you can change a lot of the settings a bit and see how it rearranges everything to cope. You've got heavy support interface set - that will recover from some droop within reason, and you can change the amount of support underneath it to find what works for you.

closed 2026-09-29 by prusaslicer-bot[bot]