prusa3d/PrusaSlicer #12996: 415: Unsupported Media Type

NiTeDK, 2024-07-02:
### Description of the bug

When I slice a file and upload and print to the printer via Prusa Link, I receive an error: HTTP 415: 415: Unsupported Media Type. Not a GCODE. This seems to be caused by an apostrophe (') in the stl filename - when I remove the apostrophe, it uploads and prints.
[image]

### Project file & How to reproduce

[Mustang GT 500 'Eleanor Kit Card set with separate front - tires.zip](https://github.com/user-attachments/files/16066407/Mustang.GT.500.Eleanor.Kit.Card.set.with.separate.front.-.tires.zip)
1. Slice
2. CTRL+SHIFT+G
3. Upload and Print

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0+win64

### Operating system

Windows 11 BUILD 22631

### Printer model

Prusa MK4

gitley, 2024-07-18:
Seems to also be caused by long filenames with spaces or hyphens. 

`Horizonal Mount Minimal - LEGO Technic 42141 McLaren Formula 1 F1 Race Car_0.4n_0.2mm_PLA_XLIS_1h26m.bgcode`

TD-er, 2024-08-12:
@gitley 
I encountered the same issue here.
Saving the file, renaming it to something shorter and then uploading did indeed fix my issue.
Thanks!

AndreasAusD, 2025-02-25:
Prusa Slicer 2.9.0 on Linux
just ran into the same problem
Prusa Slicer should know how long the name on a Prusa MK4S printer can be and keep it as short as needed
106 characters including suffix does not work, 80 does

Spreadcat, 2025-11-08:
I ran into this issue today and a) it explained the erratic upload behavior I had for a while now and b) it seemed to be resolved when I truncate the filenames to 80 characters at mentioned above.
This seems like a low hanging fruit to me for PrusaSlicer to fix.

closed 2026-09-29 by prusaslicer-bot[bot]