prusa3d/PrusaSlicer #12997: i set 1 Perimeter and slices with 2

JorgeMALopes, 2024-07-02:
### Description of the bug

I changed the settings so i have only one perimeter line 
[image]
and it slices with 2 perimeter lines
[image]

### Project file & How to reproduce

[20mm_hollow_cube.zip](https://github.com/user-attachments/files/16067000/20mm_hollow_cube.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

windows 10

### Printer model

prusa XL

u89djt, 2024-07-02:
Hi there, fellow user here. The number of perimeters is defined with respect to a surface that encloses the solid object being printed. You know that the cube as a whole is hollow, but while printing the walls, that wall is solid, and the inside surface from your perspective is an outside surface from the slicer's perspective. So, it gets a perimeter for both outside regions of the solid volume being printed, one of which is known to you as the inside of your cube. The slicer doesn't know what solid and hollow cubes are. It just knows there's object here, and no object there. If you want a single perimeter hollow cube, print what seems to you to be a solid cube with a single perimeter and no infill. Obviously you might care about the top surface, so maybe use lightning infill to minimize the material inside, and that inside this time will be from your perspective and the slicer's perspective.

JorgeMALopes, 2024-07-02:
Yeah, i just noticed that the cube was hollow. Thank you!

closed 2024-07-02 by JorgeMALopes