prusa3d/PrusaSlicer #12998: GMP not loading

Dollton, 2024-07-02:
### Description of the bug

GMP keeps timing out whenever I try to retrieve from it during the initial instillation of the PrusaSlicer

"-- Using src='https://gmplib.org/download/gmp/gmp-6.2.1.tar.bz2'
-- Retrying...
-- Using src='https://gmplib.org/download/gmp/gmp-6.2.1.tar.bz2'
-- Retry after 5 seconds (attempt #2) ...
-- Using src='https://gmplib.org/download/gmp/gmp-6.2.1.tar.bz2'
-- Retry after 5 seconds (attempt #3) ...
-- Using src='https://gmplib.org/download/gmp/gmp-6.2.1.tar.bz2'
-- Retry after 15 seconds (attempt #4) ...
-- Using src='https://gmplib.org/download/gmp/gmp-6.2.1.tar.bz2'
-- Retry after 60 seconds (attempt #5) ...
-- Using src='https://gmplib.org/download/gmp/gmp-6.2.1.tar.bz2'
CMake Error at dep_GMP-stamp/download-dep_GMP.cmake:162 (message):
  Each download failed!

    error: downloading 'https://gmplib.org/download/gmp/gmp-6.2.1.tar.bz2' failed
          status_code: 28
          status_string: "Timeout was reached"
          log:
          --- LOG BEGIN ---
            Trying 130.242.124.102:443...

  connect to 130.242.124.102 port 443 failed: Operation timed out

  Failed to connect to gmplib.org port 443 after 75340 ms: Couldn't connect
  to server

  Closing connection 0

          --- LOG END ---
          error: downloading 'https://gmplib.org/download/gmp/gmp-6.2.1.tar.bz2' failed
          status_code: 28
          status_string: "Timeout was reached"
          log:
          --- LOG BEGIN ---
            Trying 130.242.124.102:443...

  connect to 130.242.124.102 port 443 failed: Operation timed out

  Failed to connect to gmplib.org port 443 after 75019 ms: Couldn't connect
  to server

  Closing connection 0

          --- LOG END ---
          error: downloading 'https://gmplib.org/download/gmp/gmp-6.2.1.tar.bz2' failed
          status_code: 28
          status_string: "Timeout was reached"
          log:
          --- LOG BEGIN ---
            Trying 130.242.124.102:443...

  connect to 130.242.124.102 port 443 failed: Operation timed out

  Failed to connect to gmplib.org port 443 after 75005 ms: Couldn't connect
  to server

  Closing connection 0

          --- LOG END ---
          error: downloading 'https://gmplib.org/download/gmp/gmp-6.2.1.tar.bz2' failed
          status_code: 28
          status_string: "Timeout was reached"
          log:
          --- LOG BEGIN ---
            Trying 130.242.124.102:443...

  connect to 130.242.124.102 port 443 failed: Operation timed out

  Failed to connect to gmplib.org port 443 after 75003 ms: Couldn't connect
  to server

  Closing connection 0

          --- LOG END ---
          error: downloading 'https://gmplib.org/download/gmp/gmp-6.2.1.tar.bz2' failed
          status_code: 28
          status_string: "Timeout was reached"
          log:
          --- LOG BEGIN ---
            Trying 130.242.124.102:443...

  connect to 130.242.124.102 port 443 failed: Operation timed out

  Failed to connect to gmplib.org port 443 after 75004 ms: Couldn't connect
  to server

  Closing connection 0

          --- LOG END ---
          error: downloading 'https://gmplib.org/download/gmp/gmp-6.2.1.tar.bz2' failed
          status_code: 28
          status_string: "Timeout was reached"
          log:
          --- LOG BEGIN ---
            Trying 130.242.124.102:443...

  connect to 130.242.124.102 port 443 failed: Operation timed out

  Failed to connect to gmplib.org port 443 after 75325 ms: Couldn't connect
  to server

  Closing connection 0

          --- LOG END ---
          
    

make[2]: *** [dep_GMP-prefix/src/dep_GMP-stamp/dep_GMP-download] Error 1
make[1]: *** [CMakeFiles/dep_GMP.dir/all] Error 2
make: *** [all] Error 2"

### Project file & How to reproduce

brew update
brew install cmake git gettext
brew upgrade
git clone https://github.com/prusa3d/PrusaSlicer/
cd PrusaSlicer/deps
mkdir build
cd build
cmake ..
make

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

n/a

### Operating system

MacOS

### Printer model

n/a

themanyone, 2025-03-06:
Try using a VPN. I had to use a VPN because my provider randomly blocks domains it can't figure out.

SachCZ, 2025-04-14:
Hello, go to the cmakes and change the addresses to somewhere the software is available or download it separately.

closed 2025-04-14 by SachCZ