prusa3d/PrusaSlicer #13001: Modal dialogs are inivisble (arch, wayland, v2.8.0)

jaseg, 2024-07-02:
### Description of the bug

Since updating to v2.8.0, whenever PrusaSlicer shows a modal dialog, the main window looses focus, but the dialog is never shown on screen. I can still interact with the dialog through the keyboard.

### Project file & How to reproduce

Run PrusaSlicer v2.8.0 on archlinux, under wayland. I was able to reproduce the issue with both arch's native package, and with the the flatpak from flathub.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0-linux

### Operating system

Arch Linux, running gnome on wayland

### Printer model

I was only using PrusaSlicer for STL manipulation

eliliam, 2024-07-02:
Can confirm same issue, found when trying to log in via top right "Log In" button. Also running Arch, however I am using Xorg/X11. This is running the latest 2.8.0 installed via the AUR. Same behavior using the AppImage however.

jaseg, 2024-07-03:
I just observed the same bug (main window focused, but modal dialogs invisible) in Thunderbird on the same computer. So maybe this is an upstream bug?

jaseg, 2024-07-03:
@eliliam Your screenshots look like a slightly different issue: On my machine, the modal dialogs are never even shown. I don't even see a blank window. The only trace of them existing that I could find is that gnome's screenshot tool when asked to screenshot a window reserves an empty space for the invisible dialog.

vivienm, 2024-07-03:
@eliliam Got the same problem as you, with the additional log message

```
Failed to get GBM device
```

When PrusaSlicer is started with the environment variable `WEBKIT_DISABLE_COMPOSITING_MODE=1`, the log in dialog shows up and is correctly displayed.

Using Arch + Sway + Nvidia.

jaseg, 2024-07-03:
It seems we're talking about two distinct bugs after all. @vivienm the behavior I observed where the dialogs don't appear at all does not change with `WEBKIT_DISABLE_COMPOSITING_MODE=1`.

vivienm, 2024-07-03:
Interesting. Probably worth opening a second issue then, but I can't do it ATM (at work).

lordekeen, 2024-07-11:
Works with `WEBKIT_DISABLE_DMABUF_RENDERER=1` the same thing happens with Orcaslicer, which is odd.

eliliam, 2024-07-12:
> the same thing happens with Orcaslicer, which is odd.
@lordekeen isn't Orcaslicer based on Prusaslicer?

lordekeen, 2024-07-12:
> > the same thing happens with Orcaslicer, which is odd.
> 
> @lordekeen isn't Orcaslicer based on Prusaslicer?

It is based in Bambu Studio which is based in Prusaslicer, but i didnt think they would all have the same problem with their UI

eliliam, 2024-07-12:
Since they're all based on Prusaslicer it makes sense why they'd have similar issues, especially since this appears to be an underlying deeper UI issue instead of an issue with a specific feature, as I would guess that all three slicers share most of the core UI framework.

closed 2026-09-29 by prusaslicer-bot[bot]