prusa3d/PrusaSlicer #13002: Ctrl+F regression in 2.8.0

Sierra410, 2024-07-02:
### Description of the bug

The pop-up search menu doesn't seem to have a search field anymore, so having focus on it is, essentially, useless. The window, however, steals focus every time it's created.  

To actually type in the search field, the focus has to be on the main window and on the field.  

So, either:
the search field has to be clicked twice (first click summons the pop-up window, the second returns the focus),
the focus has to be returned by alt-tabbing off and back to PrusaSlicer,
or, the worst one of all, each entered symbol has to be preceded by `Esc`, which closes the popup. Entering anything, however, results in the window appearing once again...

It is very likely that this behavior is caused by the fact that PrusaSlicer is being used under Wayland on which windows can't steal the focus arbitrarily.

The easiest solution should be just duplicating the search field to appear in the pop-up window, or, alternatively, moving the field into the pop-up window, whenever the window is visible and focused.

[prusa_slicer_2.7_ctrl_f.webm](https://github.com/prusa3d/PrusaSlicer/assets/103531199/c66a9774-3b59-4917-ad78-94df20017333)
[prusa_slicer_2.8_ctrl_f.webm](https://github.com/prusa3d/PrusaSlicer/assets/103531199/3283f47a-7da5-4f29-9834-e7f1c8be1592)

### Project file & How to reproduce

Press Ctrl+F

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Arch Linux, Gnome (Wayland)

### Printer model

All of them

mildmojo, 2024-07-03:
Same issue under Linux Mint 21, which does _not_ use Wayland. Exactly as demonstrated in the second video, complete with the weird extra typeahead search box that appears over the settings list when you try to type after Ctrl+F.

I've had moderate luck clicking the search text box to initiate search instead of Ctrl+F, then clicking it again to restore focus stolen by the search window.

Overall I'd prefer a full revert to the 2.7.x UI.

ollien, 2024-08-31:
I reproduce this consistently on XFCE+i3, definitely not a Wayland issue

closed 2026-09-29 by prusaslicer-bot[bot]