prusa3d/PrusaSlicer #13004: 2.8.0 plans far more travels than 2.7.4

whirlyknurly, 2024-07-02:
### Description of the bug

On my first large print with PrusaSlicer 2.8.0 I noticed what appears to be a lot of travel moves, so I sliced the model in 2.7.4, saved it, opened it in 2.8.0, and compared the travels - sure enough 2.7.4 plans the travels somewhat better:
[image]
[image]
This adds about 20 minutes of print time and lots of noise.

### Project file & How to reproduce

Open the attached file, slice it with default settings, display travels: [riser_xl.zip](https://github.com/user-attachments/files/16074916/riser_xl.zip)

I have adjusted a few settings to see if I could fix it, but so far haven't found anything.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+linux-x64-GTK3

### Operating system

Slackware 15.0

### Printer model

Prusa XL / 2 Tool

vrilly, 2024-07-07:
I don't know if related, but i got some irregular travel moves too that didn't happen before in 2.7.4. It doesn't make sense to travel this much on the first layer for supports. Just increases the chance of a failed first layer on badly calibrated printers, high first layer speeds and a lot of wasted time travelling

[image]

[3-piece-mini-vuvzela_1.zip](https://github.com/user-attachments/files/16118459/3-piece-mini-vuvzela_1.zip)

mrwizardtx, 2024-07-11:
I had a simple print that was basically 25mm thick grid for drawer organizer that printed logically with wu but spastic with 28 hopping all over the place so there is something to this

thomascate, 2024-07-24:
I ran into this printing honeycomb sheets for an artist.

A 232 x 200 x 14 sheet causes 9h55m of travel moves in 2.7.4, but 7d1h35m of travel moves in 2.8.0.

[image]
[image]

pmcclay, 2024-08-14:
Same as @thomascate. Not hexagons, but a similar simple grid. 2.7.4 progresses sensibly through a layer by starting at some point and making a progressively expanding region of completed cells.  2.8.0 jumps from cell to cell all over with lots of long travel moves. It seems more pessimal than merely random, but that's probably observer bias.

mgurzixo, 2024-12-03:
Same issue here.
An honeycomb grid printed with a 0.4mm nozzle with a width of the grid of 0.8mm.
The hexagons are printed in completely random order with a travel  between.
This can be seen on the first layer but it is the case for all of them.
I join the STL which could be used as a test case. It has been generated by Onshape
[Grille.zip](https://github.com/user-attachments/files/17994473/Grille.zip)

TerPuc, 2025-03-06:
Fixed in PrusaSlicer-2.9.1-alpha1

closed 2025-03-06 by TerPuc