prusa3d/PrusaSlicer #13005: Prusa Slicer 2.8.0 SSL Connectivity with Octoprint

mcstoufer, 2024-07-03:
### Description of the bug

Versions:
- Prusa Slicer: 2.8.0 (Mac-x64)
- Octoprint: 1.10.2

Attempts to send sliced files to a local LAN Octoprint server are failing with:

```
Could not connect to OctoPrint: SSL peer certificate or SSH remote key was not OK:
SSL: certificate verification failed (result: 5)
[Error 60]
Note: OctoPrint version at least 1.1.0 is required.
```
Its a self-signed cert that has been trusted in the Keychain and does not expire until next year. I'm attempting to reach it via **https://octoprint.local/** I've also verified the API key is correct. It worked just fine in prior PS versions.

I can reach the web front end just fine from seperate Chrome browser and the embedded "Physical Printer" tab. Chrome tells me the self-signed cert is all valid and its happy to use it.

I can provide both Octoprint and PrusaSlicer? logs if needed.

### Project file & How to reproduce

1. A valid SSL cert is installed on OctoPrint server and trusted in the keychain on the host for which the PrusaSlicer app is run on.
2. Attempt to create a new OctoPrint physical printer with an https:// prefix.
3. Attempt to "test" connectivity. The aforementioned error will be displayed.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0 MacOS-x64

### Operating system

Mac OS 12.7.5

### Printer model

MK4

mcstoufer, 2024-07-03:
[image]
One other odd issue: When I scan to see if the Octopi host is discoverable, it immediately finds this. However, this IP is not used by any host on the  DHCP managed LAN. How can I remove this entry?

mcstoufer, 2024-07-04:
Some Console logs when the OctoPrint host is "tested":
```
default	17:40:16.535793-0700	mDNSResponder	[R7543] DNSServiceQueryRecord(1D000, 0, <mask.hash: 'AK30i3nEo+OFQAtrqSvPhg=='>, Addr) START PID[9889](PrusaSlicer)
default	17:40:16.536300-0700	mDNSResponder	[R7544] DNSServiceQueryRecord(1D000, 0, <mask.hash: 'AK30i3nEo+OFQAtrqSvPhg=='>, AAAA) START PID[9889](PrusaSlicer)
default	17:40:16.537105-0700	mDNSResponder	[R7543] DNSServiceQueryRecord(1D000, 0, <mask.hash: '06OmF/Gr7w3oY65CBf3IAA=='>, Addr) STOP PID[9889](PrusaSlicer)
default	17:40:16.537163-0700	mDNSResponder	[R7544] DNSServiceQueryRecord(1D000, 0, <mask.hash: '06OmF/Gr7w3oY65CBf3IAA=='>, AAAA) STOP PID[9889](PrusaSlicer)
default	17:40:16.585259-0700	PrusaSlicer	Trust evaluate failure: [leaf AnchorTrusted]
```

vonProteus, 2024-07-08:
same problem on Version 2.8.0+MacOS-arm64 and macOS 14.5 (23F79)
i have local custom ca trusted on computer

markphelan, 2024-07-08:
I'm having the same issue. Despite setting https://octopi.local set as the URL, and the SSL cert being trusted locally (works fine in browser) Octoprint throws an error when connecting to the pi. It seems to be resolving the DNS name and connecting via the IP address rather than hostname:
```
[error]   OctoPrint: Error getting version at https://192.168.0.89/api/version : SSL peer certificate or SSH remote key was not OK:
schannel: SNI or certificate check failed: SEC_E_WRONG_PRINCIPAL (0x80090322) - The target principal name is incorrect.
[Error 60], HTTP 0, body: ``
```
[image]

If I set the URL to https://192.168.0.89 then the test fails (since the IP address is not included in the cert, just the hostname).

vonProteus, 2024-07-09:
i have:
* octopi.lan
* octopi.local
* octopi
* 192.168.172.170

in cert and its fails 

it's worth mentioning i'm using
```
OctoPrint version : 1.10.2
OctoPi version    : 1.0.0
```

nvtkaszpir, 2024-07-27:
can you provide cert details, especially SAN section?

vonProteus, 2024-07-28:
[image]  
in previews version it worked

jamiekin, 2024-07-28:
I am getting the same error with version 2.8.0+MacOS-arm64.  My certificate is for a single hostname and does not have a SAN section.

jamiekin, 2024-07-28:
This seems to be a duplicate of issue #5506. Removing the redirect line and switching to http works as a temporary workaround, but it is not a secure or permanent fix.

mcstoufer, 2024-07-29:
Here is my SAN section:
[image]
Keep in mind that is appears to be a regression. In previous versions the config and cert were rock solid.

00monter00, 2024-08-02:
Windows and 2.8.0 same problem.

icase81, 2024-08-18:
MacOS 14.5 and 2.8.0 with the same issue. Worked in 2.7.x

theflanman, 2024-12-12:
Windows 10, 2.8.0, also having this problem.  Huge pain WRT workflow.

christianpatterson, 2025-01-01:
I switched to a Let's Encrypt certificate as a workaround for this issue. This won't work for everyone, but I hope this helps someone. My OctoPrint server is not exposed to the internet, but it's domain name is a subdomain of one I have registered publicly. (e.g. if I had `example.com` registered publicly, my OctoPrint server is `octopi.internal.example.com`.) I used `certbot certonly --manual --preferred-challenges dns` to generate the certificate. (See https://eff-certbot.readthedocs.io/en/stable/using.html#manual) Still a pain to renew the cert every quarter, but hopefully less of a pain.

apfelnutzer, 2025-01-02:
Same issue here with Prusa Slicer 2.8.0 and macOS 15.1.1 - switched back to 2.7.4 until further notice :-(.

mcstoufer, 2026-09-18:
Verified connectivity is restored as of v2.9.6. Fixes made there coupled with effective refactoring of http connectivity on the OctoPrint side looks to have resolved this.

closed 2026-09-18 by mcstoufer