prusa3d/PrusaSlicer #13007: 2.8.0 UI regression - tab key doesn't navigate between settings fields

mildmojo, 2024-07-03:
### Description of the bug

The tab key no longer moves between fields in any of the print settings, filament settings, or printer settings pages. Pressing tab causes the currently-selected field to lose focus, and no amount of tabbing or shift-tabbing will bring focus back to any field. You must use the mouse to focus each field.

### Project file & How to reproduce

No project necessary.

1. Open any print settings page.
2. Place the cursor in any settings field.
3. Press Tab. Observe that the field loses focus, and focus cannot be restored to any field without using the mouse.

[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0+linux-x64-GTK3

### Operating system

Linux Mint 21 Vanessa

### Printer model

Deltaprintr

peterkrull, 2024-07-27:
This is also an issue I am experiencing on Flathub version on Linux, fedora 40 specifically. As your example shows, this is especially annoying when setting printer speeds.

```
PrusaSlicer
Version:   2.8.0
Build:     PrusaSlicer-2.8.0+flathub.org

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 6.9.10-200.fc40.x86_64 x86_64
Total RAM size [MB]: 16,475MB
OpenGL installation
GL version:   4.6.0 (4.6 (Core Profile) Mesa 24.1.3 (git-0c49f54c76))
Profile:      Core
Vendor:       Intel
Renderer:     Mesa Intel(R) UHD Graphics 620 (KBL GT2)
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
```

m10d, 2024-08-21:
same symptoms, ubuntu 22.04 using appimage (latest / 2.8.0).  The cursor should flow between text boxes.  Or even just between elements (ie several `tab` between text boxes) would be an improvement.  This is SO MUCH mouse movement.

eximmer, 2025-01-02:
Asked a friend running MSwin to test 2.9.0: tab works like it used to.

Cloned the source and compiled 2.9.0 under Debian Stable: tab fails to
move between fields.

Installed flatpak and the new 2.9.0 version for Linux: tab fails...

A few weeks ago I had searched/checked diffs and nothing jumped out at
me so I was hoping it was a compilation compatibility issue.  I may
have time in the next week or so to try a dynamically linked version or
mess about with wxWidgets etc.

I think it will require someone much more familiar with this project to
find the solution.

ianmackinnon, 2025-05-20:
Just a possible lead for anyone investigating this bug - it may be related to the use of `wxTAB_TRAVERSAL` in various classes throughout the Slic3r GUI source (note that as well as `.cpp` files it's also used several times in the header `TopBar.hpp`).

From the (wxWidgets documentation](https://docs.wxwidgets.org/3.2/classwx_panel.html):

> ...if not all characters are being intercepted by your OnKeyDown or OnChar handler, it may be because you are using the wxTAB_TRAVERSAL style, which grabs some keypresses for use by child controls.

I've been investigating other focus-related bugs and this frequently appears to be the cause wherever the tab key is involved. The use of tab traversal in the `TopBar` class appears to not only consume the tab key event, but also switch focus to the `TopBar` panel from which very few key presses are recognised.

Possible workarounds to at least move the focus from `TopBar` include selecting the active tab using Ctrl + 1-6. This goes through the wx menu system, which appears to have precedence over the tab traversal system. It won't get you directly to the next input field but it may at least get you out of focus limbo without requiring use of the mouse.

eximmer, 2026-09-02:
This issue describes a areproducible bug.  This bug fix request should not be closed until after the bug has been fixed.

lukasmatena, 2026-09-04:
Closing as outdated. 2.x will hopefully be phased out soon, issues like this are not worth fixing. There are tab order issues in 3.x too, but those are known and in the queue.

closed 2026-09-04 by lukasmatena