prusa3d/PrusaSlicer #13008: User specified Color changes by height are corrupted when slicing for Mk4 printer

RonHuebner, 2024-07-03:
### Description of the bug

The slicer breaks the user specified color changes by height when using the Mk4 printer options. This works correctly for the Mk3 printers that I have tested.
Steps to reproduce
1. Open slicer (Version 2.8.0+win64), load STL file, slice for Mk4 printer (all variations had this bug).
2. Specify 7 color changes by height on a print. The image below shows user intended changes. These are rendered correctly.
[image]

3. Re-slice, color changes are now corrupted: missing changes and wrong colors. See image.
[image]

### Project file & How to reproduce

[Slicer issue.zip](https://github.com/user-attachments/files/16087270/Slicer.issue.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+win64

### Operating system

Microsoft Windows Pro version 10.0.22631 Build 22631

### Printer model

Prusa Mark 4 and i3 Mk3S+

gpapathanasiou, 2024-07-09:
The same with me. Tested with one color change. Only white color can be seen after slicing.
In the printer next color change is not available, but the printer stops and the color change is carried out as it should.
[image]

YuSanka, 2024-07-10:
Thanks for your report.
SPE-2410

gpapathanasiou, 2024-07-10:
[TINISTL gp.zip](https://github.com/user-attachments/files/16164036/TINISTL.gp.zip)

taz43mca, 2024-07-10:
The filament changes show in 2.8.0 Slicer and work while printing on my Mk2.5S. However, in previous slicer versions, my printer LED it showed both time remaining in printing XXXXR and time to next changes as XXXC. Items sliced using 2.8.0 only show time remaining in printing XXXR.

RonHuebner, 2024-08-25:
By turning off the generation of binary gcode, I was able to see that a "missing" color change from the Slicer interface was actually put into the gcode. It looks like the software bug is in the code that renders the Slicer image and color change chart after the user slices the print and generates the gcode/bgcode.

lukasmatena, 2024-09-12:
Fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases). Closing.

closed 2024-09-12 by lukasmatena