prusa3d/PrusaSlicer #13010: Deretraction extra length proportional to distance traveled

HippoDan, 2024-07-03:
I've been printing a lot of RC planes out of foaming LW-PLA.  That stuff oozes terribly.  I frequently need extra on deretract to make up for the amount lost in ooze while traveling.  However, for short travels it's often too much resulting in a blob at the start, or too short for long travels resulting in holes or a thinner wall.

I would love the option to put in a deretraction extra length value that will scale with the distance travelled between retract and deretract.

I would either set it up for some sort of scaling factor that would be input next to extra length amount, or have an extra extrusion length set next to the expected travel length for that amount and let the slicer scale it up or down from there.

I've tried putting the pieces really close together so they don't need retracts, but some parts are large in themselves and have retracts regardless.

This problem happens to me even with regular PLA, and although it's very subtle I think this feature would have value beyond just foaming LW-PLA.

HippoDan, 2024-07-05:
Foaming filament is independent of moisture. The foaming is a desired feature.  My filament is dry.  The oozing of regular PLA on a 150mm move is insignificant, but noticeable.    The real issue is the basic nature of lightweight filament.  It continues to expand as long as it is at temperature, so it will ooze out no matter how far you retract.  The oozing is a fact of life for LW-PLA.  The underextrusion after a long move is the problem I'm trying to solve.  One blanket extra deretract value doesn't work well as the amount needed on a 10mm move is much less than that needed after a 100mm move.   When printing resumes after the move, the nozzle is empty and the extra amount is a necessity to avoid underextrusion.

A way to account for that distance difference is what I'm looking for.

u89djt, 2024-07-05:
Aha, got you, thanks!
I suppose arraying wipe walls around/between the parts would get expensive in time and filament. And guaranteeing that those walls are used in the order you want would need some finagling. I noticed an oppotunity to impose print order by massaging layer heights - I don't know if that could do anything useful here? https://github.com/prusa3d/PrusaSlicer/issues/13021#issuecomment-2210703274
I don't know if you want to/can share a project file where you meet the problem - there are a bunch of second heads around that might spot a workaround.

HippoDan, 2024-07-15:
I'll see if I can find the most blatant example, or I'll just draw something up where I know it will. happen.  Thanks.

u89djt, 2024-07-15:
Maybe include a model that you'd normally want to print as well as extreme things so that a range of circumstances that might be met in practice can be seen?

closed 2026-09-29 by prusaslicer-bot[bot]