prusa3d/PrusaSlicer #13013: Keyboard shortcuts broken in 2.8.0

feldgendler, 2024-07-04:
### Description of the bug

Many of the keyboard shortcuts stopped working after upgrading from 2.7.4. Some things still work, a lot don't.

Ctrl-Q (exit) doesn't work.
In Print Settings tab, Ctrl-1 to switch to Plater doesn't work unless the mouse is over the main menu.
? to show the keyboard shortcuts doesn't work, ironically.
Ctrl-O works.
Alt-F to open the File menu doesn't work.

All this feels pretty random.

### Project file & How to reproduce

Just press the various keyboard shortcuts.

I'm using the PrusaSlicer-2.8.0+linux-x64-GTK3-202406271021-Ubuntu-24-04.AppImage. I used to use extracted tarballs but that option isn't available anymore.

The window manager is Awesome.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0+linux-x64-GTK3

### Operating system

Ubuntu 24.04

### Printer model

Voron 2.4 (Stealthburner)

wohltat, 2024-08-07:
I can confirm that. 

The shortcuts from menu 'Help' -> 'Keyboard Shortcuts - ?' seem to work, but not for closing, Ctrl + Q like it is shown in the dropdown menu.

-------
I use same version, also on linux.

m10d, 2024-08-21:
I came to find/submit a bug on `<tab>` navigation not working, which causes a really incredible amount of unnecessary hand & mouse movement. Looks like this has been noted numerous times (I only searched the first four pages of results for "tab"), although I cannot find any response from any developers :(

- the `alt-<menu>` is a significant accessibility issue, with another user (who is blind) noting something similar in https://github.com/prusa3d/PrusaSlicer/issues/12021
- several of the other issue have seen specific issues added, such as https://github.com/prusa3d/PrusaSlicer/issues/13002, https://github.com/prusa3d/PrusaSlicer/issues/13007, ...

ianmackinnon, 2025-05-20:
@feldgendler @wohltat I'm investigating keyboard- and focus-related bugs. Could you let me know whether these bugs are still present for you? In particular, it would be helpful to know whether Ctrl-Q works:
 
-   If you launch PrusaSlicer without letting the mouse cursor enter the window, and
-   If you launch PrusaSlicer and then move the mouse over the Plater 3D view

closed 2026-09-29 by prusaslicer-bot[bot]