prusa3d/PrusaSlicer #13015: Strange pattern

vonrudenw, 2024-07-04:
### Description of the bug

[image]

### Project file & How to reproduce

[Clamps.3mf.zip](https://github.com/user-attachments/files/16098007/Clamps.3mf.zip)
[STLs+gCode files.zip](https://github.com/user-attachments/files/16098056/STLs%2BgCode.files.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+MacOS-x64

### Operating system

MacOS Sonoma 14.5

### Printer model

MK4 Input Shaper

u89djt, 2024-07-04:
Hi there, here's why that's happening. I don't know how much detail you want, so here's most of it.
The object is overall 5.9mm high.
The print layer height is 0.2mm, which doesn't divide into 5.9.
That means that the object has to top out at 5.8mm or 6mm from the print bed.
The meshes in the project file are wobbly at the scale of about 0.000001 mm.
[image]
[image]
That matters when Z coordinates lie on a rounding up or down boundary for the available layer heights.
Points either side of the boundary of the dip on the upper surface of the object lie either side of z=5.9mm
5.90000053 mm rounds up to 6mm. 5.89999979 mm rounds down to 5.8mm.
[image]
The bottom of the hex apertures is at 2.9mm +/- 0.000001 mm, so layers have to coincide with 2.9 as well as 5.9.
[image]
Why can't the slicer recognize that it's a basically flat surface and pick one height or the other? The slicer respects your choice of layer height and tries its best to respect the model you provide within that constraint. The slicer ends up picking a single height for a slightly wobbly top if you make that height available to it. You can do that by changing some layer heights so that the top coincides with a whole layer.
[image]
Here I've set a single different layer height of 0.1 from 0.8 to 0.9mm. That shifts everything above it down by 0.1mm, and the two x.9 surfaces emerge flat in the print.
(If 5.9 was the only off-layer-period surface, we'd make a shorter perimeter shallower to add less print time, probably 4.6 to 4.7 to avoid various extra infill features.)
You can also use the "variable layer height" function to have the slicer work out where to print layers for itself. I haven't tested this extensively yet, but I assume one of the major intents of its implementation is to put surfaces where they should be. It works the way we need for this model, also introducing some finer layers for the top of the arch.
[image]
[image]
Then at the outer limits of what you might care about, I tried re-levelling the objects in the slicer with place on face. The result with layer management has more distortion, but is fixed by either of the interventions I describe above.
[image]
Conclusion? The slicer tries to do what you tell it to do :)

vonrudenw, 2024-07-04:
Hi Dave,

thanks a lot for your detailed answer. I my case, the result was surely good enough for what I needed. I was only curious, since one of the two came out OK, the other had the effect.
When critical, I will make sure that the overall thickness is a multiple of the layer height or the last layers are printed on a finer thickness.

I just wanted to let you know, in a case there could have been a bug.

Apart from that, I admire the way the Slicer works, simply great. I love the way it handle bridges and overhangs by slowing down. It has definitely improved a lot over the past 5 years and the input shaper saves a me lot of printing time.

Most of my pieces I can print without supports, another great feature.

Congratulations to very good products, both the printers (I have a mini, a MK3S and the MK4) and the software.

All the best, Wolfgang

------------
Dr. Wolfgang von Rüden
Straßergasse 17/1
1190 Wien
Austria
mob: +43 660 3787866
email: ***@***.***
www.repairteam-wien.at

image.png (view on web) <https://github.com/prusa3d/PrusaSlicer/assets/75209005/6e7848b6-7377-4bd6-beb3-d1f790bea724>
image-1.png (view on web) <https://github.com/prusa3d/PrusaSlicer/assets/75209005/982e8ce6-7af6-4471-8b37-a0690cdc794b>
image.png (view on web) <https://github.com/prusa3d/PrusaSlicer/assets/75209005/e4b70543-b563-4ce9-9d19-92f0ff7ddae5>
image-1.png (view on web) <https://github.com/prusa3d/PrusaSlicer/assets/75209005/4e64267a-3c67-4313-8295-ba152751addd>
image.png (view on web) <https://github.com/prusa3d/PrusaSlicer/assets/75209005/cc61fe7d-9975-4e3a-ae4c-618a1c1ec9e4>
image.png (view on web) <https://github.com/prusa3d/PrusaSlicer/assets/75209005/97184006-d520-4be4-945a-d08b6f777435>
image-1.png (view on web) <https://github.com/prusa3d/PrusaSlicer/assets/75209005/baa90840-f0f8-4def-b6bf-55da2b7ddf91>
image.png (view on web) <https://github.com/prusa3d/PrusaSlicer/assets/75209005/2abaf3fb-22ab-435e-a181-3cfbe0834dfa>

u89djt, 2024-07-04:
Oh hey, so I must get back in the habit of pointing out that I'm just another user checking in on github to see what issues I might meet when I need to print. I'm not affiliated with Prusa. Still, I think the Prusa folk will enjoy reading your praise - it's all for them and I wholeheartedly echo it! I enjoy explaining stuff when I don't have to mark the work, and I assume it saves the devs and support some time to have bits and pieces resolvable at the user level like this dealt with for them.

vonrudenw, 2024-07-04:
Thanks for your explanation. It is good that folks help each other like you do.

------------
Dr. Wolfgang von Rüden
Straßergasse 17/1
1190 Wien
Austria
mob: +43 660 3787866
email: ***@***.***
www.repairteam-wien.at

closed 2026-09-29 by prusaslicer-bot[bot]