prusa3d/PrusaSlicer #13017: Do not move and object after cutting

LauraNicolas, 2024-07-04:
**Is your feature request related to a problem? Please describe.**
I usually print big objects, so I use the cut feature to make parts that fit in the print bed.

After the cut, I have the option to 
1. Keep Orientation
2. Place on cut
3. Flip upside down

I always use the Keep Orientation one. 
Sometimes it also keeps the current position, sometimes it re-arranges the resulting objects.
This is a problem for me, as in case of almost identical parts after cutting, it is super difficult for me to figure out which one is which.
In any case, I always have to re-arrange the parts manually to their original position.

In the example bellow, I've done a vertical cut. The resulting A and B peaces did not move from their original position. Great, as this is what I need to fit my workflow.
[image]

Then I went ahead and cut the B part some more.
[image]

Part D did not move from it's original position, but object C got re-arranged: it's changed position and it got rotated.
[image]

**Describe the solution you'd like**
I would like two things to happen:
1. To be able to give the new object a name
2. To select an option so that the new object keeps it's orientation, rotation and position. (so, no re-arranging)

**Describe how it would work**
Explained in the image

[image]

**Describe alternatives you've considered**
I manually re-arrange after the cut
I manually re-name after the cut
When the resulting objects are too similar and it's difficult for me to know which one is which, I go and make the cut in an outside software. The problem is that if I have connectors, the export to stl does not export the connector hole/plug at all, so... I am stuck in PrusaSlicer and have to take my best guess.

**Additional context**
My guess is that when the resulting part is bigger than the build plate, it does not get re-arranged.
When the resulting part is smaller that the build plate, it gets re-arranged.

u89djt, 2024-07-05:
Fellow user passing through with a workaround while you're waiting for more options in the cut dialog: use Cut into: Parts and then arrange them. You can cut an object several times into parts and everything stays where it was in space. You can then move them around while they're part of the object by selecting them in the list on the right, or immediately split into objects, and anything suspended above the bed will just drop to the bed without moving horizontally or rotating.
[image]
[image]
[image]

LauraNicolas, 2024-07-06:
Thanks @u89djt, your tip upgraded my workflow!
PrusaSlicer team, even if this workaround is better I would love to see the request implemented. 👍 

For the top parts falling into the bed, I would love if PrusaSlicer allowed us to keep them up, but that's another topic jejeje

u89djt, 2024-07-06:
@LauraNicolas you'll be pleased to learn that there's a simple trick that lets you keep objects up in the air - just add a part and add settings to that part of zero infill and perimeters, which will sit on the bed, and you can move the main part wherever you like :)
[Shape-Box.zip](https://github.com/user-attachments/files/16116860/Shape-Box.zip)
[image]
[image]
[image]
[image]

dmak, 2024-10-15:
I have a similar issue. Would be great if Prusa does not drop any of resulting model cuts to the table, neither tries to fit them on table. In my case I would like to make vertical and then horizontal cuts which is pretty odd when parts are jumping around.

another-daniel, 2025-09-03:
Another possible way to accomplish this: add another radio button "keep in place" which preserves the orientation and the physical xyz location on the bed. For my use case, I created my own sacrificial supports. I wanted to check whether they worked without printing the whole design, so I cut just the section that I wanted to test. Sometimes, the location of the cut section on the bed has to be the same so that PS would generate the exact same support structure, otherwise the test would be useless (e.g., the placement of the object next to other ones or at the edge of the bed).1

Before:

[image]

After:

[image]

It kept the orientations, even the relative orientations, but moved the object. (@u89djt in my example, it moves the cut items even when I select parts (PS 2.9.2), as you can see above.)

MicahDoesStuff, 2026-02-02:
"Me too"
Keep Orientation is broken. It doesn't keep orientation...

dmak, 2026-09-20:
From my perspective the issue is still present in v2.9.6, because when I select "Keep orientation" I expect the split parts to be on the same positions, and not dropped to the table.

[image]

MicahDoesStuff, 2026-09-20:
This issue happens because after the cut, the current arrange bed settings are used even tho there is no indication those settings are used for cut.