prusa3d/PrusaSlicer #13019: Negative inputs for object position and rotation not accepted in 2.8

3dPrinting1127, 2024-07-04:
### Description of the bug

In previous versions, negative inputs for the object's position and rotation were allowed, but now in 2.8, it won't allow me to insert a "-" sign into these parameters.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Prusaslicer 2.8

### Operating system

Windows 11

### Printer model

MK3S+

neophyl, 2024-07-04:
Please search for existing issues before opening new ones as the instructions ask you to.  https://github.com/prusa3d/PrusaSlicer/issues/12988

3dPrinting1127, 2024-07-04:
Yes I can read thank you, I searched for this issue and nothing popped up.  Oh well

lukasmatena, 2024-07-10:
Closing in favor of #13043, which contains a list of all the duplicates.

closed 2024-07-10 by lukasmatena