prusa3d/PrusaSlicer #13021: Objects no longer get printed in the set order

KonradStuewe, 2024-07-04:
### Description of the bug

In previous versions of PrusaSlicer(I believe 2.7.4), the objects were printed in the order they were listed on the right.

This is no longer the case. It now seems to prioritize having the shortest possible travel between the objects.

While the idea is good to speed up the print by making the travel moves as short as possible, it also makes it quite a bit more unpredictable. To get the order I want, I now have to trick the slicer into choosing the right path.

If possible, I'd like to be able to go back to the old behavior.

Expected Print order:
[image]

Actual print order:
[image]

### Project file & How to reproduce

[Print Order.zip](https://github.com/user-attachments/files/16102694/Print.Order.zip)

1. Add multiple objects to the build plate.
2. Sort the objects in the list in the required order.
3. Arrange the objects in a line on the built plate, while giving them a different order.
[image]
5. Slice the objects.
6. _The order in which the objects get printed is now different from the list._
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0+win64

### Operating system

Windows 10 22H2 19045.4529

### Printer model

Prusa MK4 Kit

u89djt, 2024-07-05:
Pausing to note that I'm a fellow user, not affiliated with Prusa, I wonder; what is the advantage in printing the objects in the same order as the list when you're not performing sequential printing?

KonradStuewe, 2024-07-05:
> Pausing to note that I'm a fellow user, not affiliated with Prusa, I wonder; what is the advantage in printing the objects in the same order as the list when you're not performing sequential printing?

I use it mostly to add my own prime tower when I do a filament swap.
The nozzle tends to ooze a bit after making a filament change, so I add a little cube to clean the nozzle and give it a chance to properly pressurize.

Right now, I have to keep moving the prime tower around the build plate in the hopes that it actually gets printed first

u89djt, 2024-07-05:
I think I have a workaround for you. [wipe at 3p8 Print Order.zip](https://github.com/user-attachments/files/16109001/wipe.at.3p8.Print.Order.zip)
[image]
[wipe at 3p8 Print Order.zip](https://github.com/user-attachments/files/16109001/wipe.at.3p8.Print.Order.zip)
Add a layer height setting to the wipe object to be lower than the other objects at the level you want to wipe at so that it is extruded first because the stuff around it is higher.
You could re-align them above that level to allow the slicer to optimize travel as well as possible while the print progresses up to the next filament change, or indeed stop the wipe tower as soon as it's done its job if there's only one change.

fe60, 2024-07-06:
> In previous versions of PrusaSlicer(I believe 2.7.4), the objects were printed in the order they were listed on the right.

@KonradStuewe Are you sure about that?<br />I found at least two feature requests asking for this behaviour (Nov 21 #7318, Nov 20 #5175).

closed 2026-09-29 by prusaslicer-bot[bot]