prusa3d/PrusaSlicer #13024: Physical Printer Tab is blank with all printers.

bobbyh41, 2024-07-05:
### Description of the bug

The prusa connect tab works, but also goes blank after some time, but the physical printer tab never shows anything with my Prusa XL or my MK4. And I do have it logged in to send files over my network.
[image]

### Project file & How to reproduce

I have never seen it work.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Mac 14.5 M1 IMac

### Printer model

Prusa XL and Prusa MK4

buster1407, 2024-07-06:
Having the same issue with my Prusa Mini+ and a Macbook Air M1.
With the same printer and a Windows PC everything works fine.

sshockley, 2024-07-10:
I'm seeing the same issue on PS 2.8.0 on Windows.  I am not using any of the "connect" features and prusaslicer.exe is blocked from accessing the Internet.

ThatRed, 2024-07-21:
I had to reinsert the API Key to make it work: Plater, right side, Printer, click cog, "Edit physical printer", reinsert the API Key, click "OK". Make sure that the API Key actually works, test it on the Prusa Link website....

TCPhelps, 2024-10-11:
I also have this issue with the Prusa XL. It showed the screen on an earlier version of PrusaSlicer, but not on the current one. I can still log into the XL using Chrome.

Here's the copied info from PrusaSlicer's 'System Info' section:

PrusaSlicer
Version:   2.8.1+win64
Build:     PrusaSlicer-2.8.1+win64-202409181359

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19045), 64-bit edition
Total RAM size [MB]: 16,872MB
OpenGL installation
GL version:   4.6.0 (4.6.0 - Build 32.0.101.5768)
Profile:      Core
Vendor:       Intel
Renderer:     Intel(R) Iris(R) Xe Graphics
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_3DFX_texture_compression_FXT1
GL_AMD_depth_clamp_separate
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
[... 173 lines cut ...]
GL_KHR_debug
GL_KHR_no_error
GL_KHR_shader_subgroup
GL_KHR_shader_subgroup_arithmetic
GL_KHR_shader_subgroup_ballot
GL_KHR_shader_subgroup_basic
GL_KHR_shader_subgroup_clustered
GL_KHR_shader_subgroup_quad
GL_KHR_shader_subgroup_shuffle
GL_KHR_shader_subgroup_shuffle_relative
GL_KHR_shader_subgroup_vote
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_timeline_semaphore
GL_OVR_multiview
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

Tedderouni, 2024-12-12:
I'm experiencing a similar issue.

I'm on Version 2.8.1+MacOS-x64 on a 2017 Macbook Pro (Intel) running MacOS Ventura 13.7.1.

In my case, I was using HTTP Digest authentication with multiple physical printer profiles for different nozzle sizes on a Prusa MK4, and it always worked fine. But this week I upgraded my MK4 to a MK4S and created new profiles, and I had this issue with the new profiles, even though the old ones continued to work. I especially had this issue if I started PrusaSlicer with the new profile selected (after previously closing it with the new profile selected), and sometimes it would start to work if I switched to an old MK4 profile and then switched back to the MK4S profile.

What eventually worked for me was switching (or recreating) my physical printer profiles to use API Key authorization instead of HTTP Digest. After doing that, it works consistently so far.

cb88, 2025-03-04:
2.9.0 + latest 6.2.1 Firmware on Prusa Mini still does this, if I click open physical printer URL it opens externally in browser and I can open it in Firefox and it works there so its something with the embedded browser.

cb88, 2026-09-02:
Great you labeled this as a legacy issue bot. Good job... but did it get fixed? it seems like a simple configuration issue with the embedded browser.

sshockley, 2026-09-03:
> Great you labeled this as a legacy issue bot. Good job... but did it get fixed? it seems like a simple configuration issue with the embedded browser.

idk I tried to repro it in the 3.0.0 alpha but I can't even finish the first start wizard without a connection or a Prusa printer.  PS 2.9.6 shows "Something went wrong" for me which I guess is... better?