prusa3d/PrusaSlicer #13028: Island detection for resin supports

kozross, 2024-07-05:
**Is your feature request related to a problem? Please describe.**

When doing manual resin supports, it is important to know where _islands_ (that is, overhangs not connected to anything in the same layer) are on a model, as these are the most important elements that require supporting. Additionally, automatic supports, while they will pick these up, are often excessive for what is needed, especially on complex models, such as miniatures. 

At least one competing product (Lychee) has automatic island detection, which highlights islands on a model. You can see how this looks below: the red section is an island, that will need supporting:

[image]

Prusa Slicer currently has no such feature. This makes manual supports a chore: I have to place a support, slice, scrobble up through the layers until I spot an island (_if_ I spot it...), go back, place a support (and hope it's in the right place...), reslice, etc. If island detection existed, it would make this process a thousand times less tedious, to say nothing of faster.

**Describe the solution you'd like**

Island detection similar to what Lychee does currently. This could work similarly to how overhang detection works for FDM supports: you specify a minimal size of island you're interested in, and they get highlighted when doing manual supports.

**Describe how it would work**

This should be similar to how FDM overhang highlighting currently works. In the manual support mode for resin prints, you would choose to highlight islands, as well as specifying a minimum island size you care about. This would then show the islands in red on the model. This is quite similar to how Lychee does it now.

**Describe alternatives you've considered**

The only alternatives right now are as follows:

* Use automatic supports; and
* The process I just described.

Automatic supports are usually excessive, adding a lot of material use and scarring for no real benefit. This is not unique to PrusaSlicer: practically _all_ resin slicer auto-supports are discouraged for complex models such as miniatures for exactly the reason described. The workflow I described above is slow, tedious and error-prone. Neither of these situations are fantastic.

**Additional context**

PrusaSlicer is currently the _only_ open-source resin slicing option that works, and having such a feature would make it much easier for people who don't want to use proprietary products, which constantly break, paywall features or generally make your life miserable if you're not on Windows.

atiniir, 2024-12-22:
I'd really like to see this feature added, or at least know if it's on the roadmap. I'm likely to stop using my SL1S until this is added as it's so much more fiddly, time consuming and risky to use compared to Saturn + Chitubox

mnemnion, 2025-02-11:
This is a great idea. I wanted to toss an idea out there: UVTools has best-in-class island detection already, so implementing this feature might be as 'simple' as exporting that data to PrusaSlicer and displaying it on the model.  

I've begun looking into the UVTools side of that design actually, the learning curve is fairly steep so I can't even guess how far I might get with it.

closed 2026-09-29 by prusaslicer-bot[bot]