prusa3d/PrusaSlicer #13030: New CLI option --material-profile gets ignored for single-material printers.

schiele, 2024-07-05:
### Description of the bug

The new option --material-profile works with multi-material printers but gets silently ignored with single-material printers.

For the following examples, I set the filament to "Prusament ASA" in the GUI and quit the GUI. After that I tried the following commands on the command line.

Examples with "Original Prusa i3 MK3S & MK3S+ MMU2S" profile where the behavior is correct:
```
$ prusa-slicer -g --printer-profile "Original Prusa i3 MK3S & MK3S+ MMU2S" --print-profile "0.20mm QUALITY @MK3" --material-profile "Prusament PLA @MMU" cube.stl
10 => Processing triangulated mesh
20 => Generating perimeters
30 => Preparing infill
45 => Making infill
65 => Searching support spots
69 => Alert if supports needed
88 => Estimating curled extrusions
89 => Calculating overhanging perimeters
88 => Generating skirt and brim
90 => Exporting G-code to cube_0.2mm_{printing_filament_types}_MK3SMMU2S_{print_time}.gcode
Slicing result exported to cube_0.2mm_PLA_MK3SMMU2S_8m.gcode
```
Uses PLA as expected.

```
$ prusa-slicer -g --printer-profile "Original Prusa i3 MK3S & MK3S+ MMU2S" --print-profile "0.20mm QUALITY @MK3" --material-profile "Prusament PETG @MMU" cube.stl
10 => Processing triangulated mesh
20 => Generating perimeters
30 => Preparing infill
45 => Making infill
65 => Searching support spots
69 => Alert if supports needed
88 => Estimating curled extrusions
89 => Calculating overhanging perimeters
88 => Generating skirt and brim
90 => Exporting G-code to cube_0.2mm_{printing_filament_types}_MK3SMMU2S_{print_time}.gcode
Slicing result exported to cube_0.2mm_PETG_MK3SMMU2S_9m.gcode
```
Uses PETG as expected.

Now we try the same with a single-material printer:
```
$ prusa-slicer -g --printer-profile "Original Prusa i3 MK3S & MK3S+" --print-profile "0.20mm QUALITY @MK3" --material-profile "Prusament PLA" cube.stl
10 => Processing triangulated mesh
20 => Generating perimeters
30 => Preparing infill
45 => Making infill
65 => Searching support spots
69 => Alert if supports needed
88 => Estimating curled extrusions
89 => Calculating overhanging perimeters
88 => Generating skirt and brim
90 => Exporting G-code to cube_0.2mm_{printing_filament_types}_MK3S_{print_time}.gcode
Slicing result exported to cube_0.2mm_ASA_MK3S_8m.gcode
```
Here it silently ignores my selection of PLA and uses the ASA setting from the previous GUI session instead.

```
$ prusa-slicer -g --printer-profile "Original Prusa i3 MK3S & MK3S+" --print-profile "0.20mm QUALITY @MK3" --material-profile "Prusament PETG" cube.stl
10 => Processing triangulated mesh
20 => Generating perimeters
30 => Preparing infill
45 => Making infill
65 => Searching support spots
69 => Alert if supports needed
88 => Estimating curled extrusions
89 => Calculating overhanging perimeters
88 => Generating skirt and brim
90 => Exporting G-code to cube_0.2mm_{printing_filament_types}_MK3S_{print_time}.gcode
Slicing result exported to cube_0.2mm_ASA_MK3S_8m.gcode
```
Again my selection of PETG gets ignored in favor of the previously selected ASA setting in the GUI.

I did some basic debugging and found that it still feeds the correct string (the supplied one) to the function call in line 551 of `src/libslic3r/ProfilesSharingUtils.cpp`
```
            extruders_filaments.emplace_back(ExtruderFilaments(&preset_bundle.filaments, i, material_preset_names[i]));
```
This does not cause the desired effect on single-material printers though. I didn't dig deeper since I was not familiar enough with the code.

### Project file & How to reproduce

To reproduce the problem use any STL file on the command lines above. The actual content of the STL file is completely irrelevant to this problem. In my example, I used an STL file containing a trivial cube of 1cm in size.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Clear Linux OS 41990

### Printer model

Original Prusa i3 MK3S & MK3S+

PrivateButts, 2024-07-31:
Ran into this as well. I've been working around it by messing with the PrusaSlicer.ini's filament settings. Here's my python snippet if it helps anyone else:

```python
with open(PRUSASLICER_INI_PATH, "r") as f:
    ini = f.read()

ini = re.sub(r"filament\s+=.+", f"filament = {material_profile}", ini)
with open(PRUSASLICER_INI_PATH, "w") as f:
    f.write(ini)
```

Sebastian1989101, 2024-11-29:
+1 just encounter this issue as well while trying to implement a API for my customers to check average cost of print jobs. Seems like workaround works at least tho.

closed 2026-09-29 by prusaslicer-bot[bot]