prusa3d/PrusaSlicer #13031: Lack of filament change time on printer display

taz43mca, 2024-07-05:
### Description of the bug

Although the printer did stop at the designated layer for the filament change designated in slicer, the time to change did not show on the printer display. previous versions showed remaining time "R" and time to filament change "C" on printer display. My printer is a MK2.5S and both firmware and software are latest versions.

### Project file & How to reproduce

[Oval Frame C 1988 A STL_0.2mm_PLA_MK2.5S_2h9m_6.09mm_$0.47.gcode.zip](https://github.com/user-attachments/files/16113345/Oval.Frame.C.1988.A.STL_0.2mm_PLA_MK2.5S_2h9m_6.09mm_.0.47.gcode.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

windows 11 Pro

### Printer model

MK 2.5S

benring, 2024-07-07:
I have experienced the exact same issue on my MK3S+. This happened recently when I upgraded from 2.74 to 2.8. I also tried upgrading the firmware on the printer from 3.13 to 3.14 and still had the same issue. I can also confirm that models sliced on 2.74 or earlier still work properly.

gpapathanasiou, 2024-07-09:
I have the same problem with you. I also discovered that, when you slice a project with filament change, the result in the preview shows only the basic color.
[image]

YuSanka, 2024-07-10:
@gpapathanasiou, I believe your issue is the same as #13008 and isn't related with this issue.

@taz43mca and @benring, provide us please the 3mf file, perhaps evaluating your configuration can help us

fe60, 2024-08-20:
This was reported for 2.8.0-alpha5 in #12849

lukasmatena, 2024-09-12:
Fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases). Closing.

sarusani, 2024-09-15:
@lukasmatena Just testet it with [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases) and it's not fixed. Still the same behavior as in 2.8.0. M73 Cxx codes are not generated for color changes.

lukasmatena, 2024-09-16:
Thanks, my bad. The issue that was fixed was #13008, which seems to be what @gpapathanasiou mentioned here, but this issue was originally about something else. Reopening.

samdurnxn, 2024-09-18:
If it can help to resolve the bug. I made some tests with PrusaSlicer 2.8.1-RC1
Add a shape box.

If I add a break(M601) and a color change(M600). In the Gcode file, no problem M73 Cx  and M73 Dy are correctly insert

But If I use multimaterial painting, Tx are correctly insert but not the M73 Cx and M73 Dy

It has been confirm by DanoPernis on the issue : https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4089#issuecomment-2358326544

I can provide my test file if you need it.

taz43mca, 2024-09-18:
I have a MK2.5S and did note that I still got the changes but they did not show up on the LCD screen as XX:XXC. All I get is the XX:XXR for the remaining time left in print.
I also had some intermittent filament change pauses occur that were not inserted in the slicer when I did my slice.Right now I am slicing with version 2.5.0 and am having no such problems. I am, however, having a problem as follows:Layer 1 - fan speed (in tune) 0Layer 2 - fan speed               84Layer 3 - fan speed              170   *Layer 4 - fan speed               255  *If I do not immediately go to TUNE and drop the fan speed at the start of layer 3 and 4 to 100 or lessI get a Thermal shutdown.  I thought that the firmware (latest) may be adjusting the fan speed for the 24 Volt hot end of the MK3 and above and not for the 12 Volt MK 2.5. The slicer has a max fan speed set at 100 so the problem has to be in the firmware.All parts of my printer are new, motherboard Rambo 1.3b, and hot end.If you send me the 2.8.0 Slicer mod I will try it and report findings.Regards,Carl

 

It has been confirm by DanoPernis on the issue : prusa3d/Prusa-Firmware-Buddy#4089 (comment)

LoakAudio, 2024-11-02:
Hi all,
This bug is present on an MK3.5, no MMU, firmware 6.1.3 & PS 2.8.1
Color Change is asked layer 26, gcode viewer show it and the print is ok, but "Prochain Changement : N/A" ("Next Change : N/A" in english I gess) is shown on the main screen during the print
Here is the bgcode used : 
[Prusa i3 MK3 Memorial Plain_0.6n_0.32mm_PLA_MK3.5_4h4m.zip](https://github.com/user-attachments/files/17607649/Prusa.i3.MK3.Memorial.Plain_0.6n_0.32mm_PLA_MK3.5_4h4m.zip)

lukasmatena, 2024-12-20:
Can you please retest this with 2.9.0-rc1? It turns out there were multiple issues with the color change time estimate.

taz43mca, 2024-12-20:
Downloaded and ran a quick test and the change time now shows up on the LCD screen. Thanks for the fix.

lukasmatena, 2024-12-20:
Thanks for the update. Closing.

benring, 2024-12-22:
Confirmed. Works for me too!

Just in time as my upgrade kit has already shipped.