prusa3d/PrusaSlicer #13033: Generated Gcode doesn't set extruder_colour when it is set to filament color

PinkPandaKatie, 2024-07-05:
### Description of the bug

If I go to Printers -> Extruder # -> Extruder Color, and click "Reset to Filament Color" for all of the extruders, then the generated Gcode contains a blank `extruder_colour` value. This causes the Filament Mapping screen to not show the colors for the filaments.

PrusaSlicer's Gcode viewer is still able to view the filaments because it looks at the `filament_colour` in the Print Metadata block, but the printer can only look at the Printer Metadata block which only contains `extruder_colour`, since Print Metadata is Deflate compressed.

[image]

### Project file & How to reproduce

1. Go to Printers -> Extruder # -> Extruder Color and click "Reset to Filament Color" for all extruders
2. Go to Filaments and set up two or more user presets with different colors.
3. Slice a multi-color print that uses these filaments.
4. Export the Gcode and examine the header. Note that `extruder_colour` is blank.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Ubuntu 22.04

### Printer model

MK4+MMU3

PinkPandaKatie, 2024-07-05:
Project file upload failed the first time. Here it is:

[screwtray-project.zip](https://github.com/user-attachments/files/16113789/screwtray-project.zip)

PinkPandaKatie, 2024-07-05:
I don't know where the best place to correct this would be. One possible solution would be to add "filament_colour" to the list of keys copied to the printer metadata block in `GCodeGenerator::_do_export`, then have the printer firmware check `filament_colour` if the corresponding `extruder_colour` is blank. It could also probably be copied to `extruder_colour` in that function, but that seems like a hack.

lordofhyphens, 2024-09-10:
It's not a code fix, but it looks like at least HappyHare's preprocessor will find the correct values if you put in a fake `extruder_colour = ` entry in your start gcode  that has a `;` separated `[filament_colour_0];[filament_color_1]...[filament_colour_n]`

closed 2026-09-29 by prusaslicer-bot[bot]