prusa3d/PrusaSlicer #13036: Can we please get the old UI back?

ScreennameTaken, 2024-07-06:
With where the File/Edit/Window/View etc buttons being where they should be.

dougemon, 2024-07-06:
I agree, while I like the new Plater/Filaments/Printers bar, the File/Edit/Window needs to be laid out as a bar above. This is the norm for basically all windows applications.

novaplusplus, 2024-07-07:
A very baffling decision. Because there wasn't enough horizontal space to fit a few extra menu items?

[image]

Please don't push to make the prusaslicer UI more like bambu studio or orca. Let it be its own more tried-and-true thing, that's literally why I still use it.

apopticon, 2024-07-07:
I don't know of anyone using Prusaslicer on a phone. Why use UI elements that are only suited for a mobile environment? Moving all the customary tool bar menus to a new button is just terrible, and is made even worse by removing the keyboard shortcuts as well. Why was this deemed important enough to warrant the time and effort??? Change for change's sake? PLEASE put the menu back, or at least make it a configurable option!

themanyone, 2024-07-08:
I made a pull request fixing my most annoying issues https://github.com/prusa3d/PrusaSlicer/pull/12994

ktmom743, 2024-07-09:
Adding a additional level of mouse navigation through a drop down menu is regressive.  Please at least add a preferences option to revert to the old UI.

redPanther, 2024-07-09:
In many linux applications menu is also squeezed into an icon, but mostly they make an option to "always show menu bar". Please Please do that to. If you want the "modern" style - ok, if you want to copy bambu - ok... BUT let us users the choice! Make an Option to enable/disable the menu bar.
BTW I like the firefox way. There you can press the ALT key and then menu is shown. This is very convenient (and of course you can always show it, if desired).

themanyone, 2024-07-12:
The button code that brings up the menu is private. So I made a public method, `TopBarItemsCtrl::TriggerMenuButtonPopup()`. There was no global keyboard event handler to intercept the Alt key to bring up the menu. So I made one, `void MainFrame::OnKeyDown(wxKeyEvent& event)`. It calls `TriggerMenuButtonPopup()` when `Alt+F` is pressed (also Alt+E, Alt+V, etc.) And a few more keyboard shortcuts for pull request https://github.com/prusa3d/PrusaSlicer/pull/12994 (I rather prefer to use the keyboard than hunt around 4 screens for my mouse.)

themanyone, 2024-12-08:
Addressing a few more odd focus and keyboard navigation issues with search bar and results window in #13691

closed 2025-05-12 by Jan-Soustruznik