prusa3d/PrusaSlicer #13037: PrusaSlicer 2.8 build errors (Windows - missing Boost)

amatulic, 2024-07-06:
### Description of the bug

Ran into unexpected errors updating my local repository to the latest PrusaSlicer 2.8. So I completely deleted my local repo and started afresh, cloning the PrusaSlicer repo and following the build instructions.

The first cmake command, `cmake ..` resulted in this error:
````
CMake Error at C:/Program Files/CMake/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Boost (missing: Boost_INCLUDE_DIR system filesystem thread
  log locale regex chrono atomic date_time iostreams nowide) (Required is at
  least version "1.66.0")
````

### Project file & How to reproduce

Follow the build instructions for Windows at https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md

The error occurs in the first `cmake` command in manual build step 2A.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.9

### Operating system

Windows 11 / Visual Studio 2022

### Printer model

Prusa MK3S/MMU2S

FloatyDev, 2024-07-13:
This also happened to me, any idea?

AmyTheCute, 2024-07-19:
Same issue here. Usually a Mac dev but I can't compile this on windows at all, following the instructions as is.

AmyTheCute, 2024-07-19:
are you running this command in the build subdirectory?

amatulic, 2024-07-20:
> are you running this command in the build subdirectory?

Yes. I followed the directions two or three times with the same result. Also I ran the automated build script, and also got an error.

amatulic, 2024-07-30:
This might be related to #12884 on Linux.
Since I reported this problem, #12884 resulted in a patch to some include files a couple weeks ago.
Trying to build it again on Windows, it seems to be working. It actually downloaded all the boost packages in the second cmake step. I couldn't get past the first cmake step before.

I'm closing this as resolved.

amatulic, 2024-07-30:
This might be related to #12884 on Linux.
Since I reported this problem, #12884 resulted in a patch to some include files a couple weeks ago.
Trying to build it again on Windows, it seems to be working. It actually downloaded all the boost packages in the second cmake step. I couldn't get past the first cmake step before.

This can be closed as "fixed" or "cannot reproduce".

amatulic, 2025-01-19:
Well, something got fixed because I've been able to build PrusaSlicer since 2.8.1. Closing.

closed 2025-01-19 by amatulic