prusa3d/PrusaSlicer #13038: insufficient memory in 2.8.0

Alsafi, 2024-07-06:
### Description of the bug

o problema é que ta aparecendo um erro falando que nao tenho espaço de memoria, mas com o gerenciador de tarefas aberto nao passou de 40% a memoria

espero que possa resolver isso logo, se nao der, vou passar pra 2.7

### Project file & How to reproduce

the error is in Portuguese
[image]
[Raiva Markit3D.zip](https://github.com/user-attachments/files/16117597/Raiva.Markit3D.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8

### Operating system

windows 10

### Printer model

creality k1

tpihl, 2024-07-08:
-1

lukasmatena, 2024-07-08:
Thanks, reproduced. We will fix it in the next version.
SPE-2402

YuSanka, 2024-07-10:
@tpihl, How did you confirm this issue? just tested attached 3mf? Or do you have your own case?

@Alsafi, How did you make your 3mf? Did you add tool changes, when your printer profile contained 6 extruder and then decrease count of them?

Alsafi, 2024-07-12:
@YuSanka to be more exact, I got a 3mf that worked and had 6 extruders, then I deleted the keychain on it and added this one from the 3mf, I then changed the colors and sliced ​​it, then I got this error, then removing the extruders and leaving 1 turn on the normal, then putting it back to 6 the error happens again.
I tested it again now and the error happens

YuSanka, 2024-07-12:
Perfect! So, we reproduced your issue correctly :) Fix will be a part of next release.

tpihl, 2024-07-13:
I use my gaming rig also for slicing and have ran out of memory several times where windows reports majority of memory free. Monstley with several complex model on the xl plate when I have copied some models to try different ways to place them (so I hide the currently non relevant copies to see what the time/usage/warnings and similar would be)

lukasmatena, 2024-09-12:
Fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases). Closing.

closed 2024-09-12 by lukasmatena