prusa3d/PrusaSlicer #13040: XY Compensation Interferes with Painted Seams

curby, 2024-07-07:
### Description of the bug

Painted seam enforcers/blockers have no effect when XY Compensation is used. Resetting XP Compensation to 0 makes seam painting work again.

### Project file & How to reproduce

Project file:
[TEST Connectors.3mf.zip](https://github.com/user-attachments/files/16120838/TEST.Connectors.3mf.zip)

Painted seams:
[image]

Seam position with 0.15mm XY Compensation
[image]

Seam position with 0 XY Compensation
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0+MacOS-arm64

### Operating system

macOS 12.7.4

### Printer model

Prusa MK4

SachCZ, 2024-07-08:
Hi I have got a followup. **It will actually be very hard to fix**. Please try to use the Shrinkage compensation in Filaments -> Advanced.

curby, 2024-07-08:
Thanks for the feedback! I'll give that a shot next time I need to do something similar, and hope for the best for an eventual solution. Cheers!

Idle musing: I imagine that painted elements are literally painted on the model's original surface, and that XY Compensation grows the outer shell "beyond the paint." If so, perhaps XY Compensation could be modeled/visualized in the editor view, and then painted elements could be applied to that?

SachCZ, 2024-07-08:
Hi, your assumptions are correct and it is sadly easier said than done :smiling_face_with_tear:

jonathanrswift, 2026-09-02:
This IS a reproducible bug, so should not be closed. Hopefully the new architecture makes it easier to fix, and otherwise there should be a tooltip explaining the issue at the very least, which would be easy to implement. The fact the tooltip was never updated on later 2.X versions when this is literally a single line text change in the code is bonkers.