prusa3d/PrusaSlicer #13041: 2.8.0 on Linux can't find STL/3MF in current dir

CrashCash, 2024-07-08:
### Description of the bug

I just installed 2.8.0 on Devuan, and it can't find STL or 3MF files unless I give the full path:

> slicer basket-left-small.3mf 
No such file: basket-left-small.3mf

> slicer /storage/3D_printer/my_designs/shower_rack/basket-left-small.3mf
[works and opens file]

### Project file & How to reproduce

Download AppImage with browser to /storage/3D_printer/prusa/prusaslicer

As root:
# cd /storage/3D_printer/prusa/prusaslicer
# chmod a+x PrusaSlicer-2.8.0+linux-x64-GTK3-202406271021-Ubuntu-24-04.AppImage
# cd /opt
# /storage/3D_printer/prusa/prusaslicer/PrusaSlicer-2.8.0+linux-x64-GTK3-202406271021-Ubuntu-24-04.AppImage --appimage-extract
# mv squashfs-root PrusaSlicer-2.8.0
# cd /usr/local/bin/
# ln -s /opt/PrusaSlicer-2.8.0/AppRun slicer

As user:
> cd /storage/3D_printer/my_designs/shower_rack
> ls *.3mf
basket-left-large.3mf  basket-left-small.3mf  basket-right-large.3mf  basket-right-small.3mf  hook.3mf  hook-short.3mf  support-end.3mf  support-mid.3mf  support-mid-short.3mf
> slicer basket-left-small.3mf 
No such file: basket-left-small.3mf

> slicer /storage/3D_printer/my_designs/shower_rack/basket-left-small.3mf
[works and opens file]

> /opt/PrusaSlicer-2.8.0/AppRun basket-left-small.3mf 
No such file: basket-left-small.3mf

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+linux-x64-GTK3

### Operating system

Devuan GNU/Linux excalibur/ceres (aka Debian w.o systemd)

### Printer model

Prusa MK4

MattBlissett, 2024-07-23:
If you use Zsh, a workaround is the `(:a)` expansion modifier.

```
PrusaSlicer-2.8.0+linux-x64-GTK3-202406271021-Ubuntu-24-04.AppImage *.stl(:a)
```

(I execute the AppImage directly.)

CrashCash, 2024-07-23:
I use bash. My workaround is to go back to 2.7.4. That also fixes the "horrible new UI" issue and lets me stick with GTK2.

closed 2026-09-29 by prusaslicer-bot[bot]