prusa3d/PrusaSlicer #13043: PrusaSlicer 2.8.0 does not accept input from keyboard numeric row (mainly `-`)

sharky98, 2024-07-08:
### Description of the bug

When entering values in a field, the negative symbol <kbd>-</kbd> (U+002D HYPHEN-MINUS character) when entered from the numeric row (scancode `0c`) is not considered by PrusaSlicer. Using the numpad substraction symbol (scancode `4a`), which is also the <kbd>-</kbd> (U+002D HYPHEN-MINUS character) symbol, does work.

So far, I was not able to use the <kbd>-</kbd>, scancode `0c` in any input (search bar, translation, rotation, settings, renaming an object, etc.).

The key <kbd>=</kbd> is also ignored in my case.

There seems to have a recurring pattern with various affected key, but the negative sign seems to be common for all reported cases I was able to find: #12939, #12988, #12985, #12961, #13019. #13023 may also be related as the user mention that the numeric row `+` does not work.

### Project file & How to reproduce

1. Focus in any text or numeric field.
2. Use the numeric row <kbd>-</kbd>.
3. See no input.

[empty-project.zip](https://github.com/user-attachments/files/16122251/empty-project.zip)

Using the on-screen keyboard to demo the behavior.

https://github.com/prusa3d/PrusaSlicer/assets/8150191/c4070fd6-091d-4a52-badb-7fce67043ca0

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+win64

### Operating system

Windows 11 & Windows 10

### Printer model

Ender 3 S1 Pro

themanyone, 2024-07-12:
I would volunteer to help. But I can not reproduce this on Linux. And do not have th' resources to test Windows.

_Although not a source of trusted code, AI is good at associating tokens to find popular ideas to try. The following excerpt from claude.ai might prove helpful._

------ claude.ai chat -------

themanyone: Any clue what might be intercepting the keyboard and preventing typing '-' key on WIN32?

claude: If the '-' key is not working on a Windows system, especially in a wxWidgets application, there could be several reasons for this. Here are some possibilities and ways to investigate:

Global Hotkeys:
Some applications or system utilities might be capturing the '-' key globally. Check if the key works in other applications.
Keyboard Layout:
Ensure that the correct keyboard layout is selected in Windows settings.
wxWidgets Event Handling:
Your application might be inadvertently capturing or stopping the propagation of the '-' key event. To investigate this:

Implement a wxEVT_CHAR_HOOK handler in your main frame or the problematic window:
```cpp
void MyFrame::OnCharHook(wxKeyEvent& event)
{
    wxLogDebug("Key pressed: %d", event.GetKeyCode());
    event.Skip();
}
```
Bind this handler:
```cpp
this->Bind(wxEVT_CHAR_HOOK, &MyFrame::OnCharHook, this);
```

    Global Hotkeys: Some applications or system utilities might be capturing the '-' key globally. Check if the key works in other applications.
    Keyboard Layout: Ensure that the correct keyboard layout is selected in Windows settings.
    wxWidgets Event Handling: Your application might be inadvertently capturing or stopping the propagation of the '-' key event. To investigate this:
        Implement a wxEVT_CHAR_HOOK handler in your main frame or the problematic window:

        cpp

void MyFrame::OnCharHook(wxKeyEvent& event)
{
    wxLogDebug("Key pressed: %d", event.GetKeyCode());
    event.Skip();
}

Bind this handler:

cpp

        this->Bind(wxEVT_CHAR_HOOK, &MyFrame::OnCharHook, this);

    This will log all key presses, allowing you to see if the '-' key event is being received.
    Focus Issues: The '-' key might not work if the focus is on a control that doesn't accept it. Ensure the focus is on the correct control when you're trying to type '-'.
    Custom Event Handlers: Check if you have any custom event handlers that might be intercepting keyboard events.
    Accelerator Tables: Review your accelerator tables to ensure '-' isn't being used as a shortcut.
    Input Method Editors (IMEs): If you're using an IME, it might be intercepting certain keystrokes. Try disabling the IME temporarily.
    Windows API Hooking: Some applications use low-level keyboard hooks. You can use tools like Process Monitor to see if any processes are hooking keyboard input.
    Accessibility Features: Certain Windows accessibility features can interfere with keyboard input. Check if any such features are enabled.
    Hardware Issues: Although unlikely if it's just the '-' key, it's worth checking if the key works in other applications or on the Windows on-screen keyboard.

To further diagnose the issue, you could implement a more detailed logging system for key events in your application. This would help you understand at what level the key press is being lost or intercepted.

If you can provide more context about where in your application this is occurring (e.g., in a specific text control, globally across the app), I can offer more targeted advice over at claude.ai

Dycus, 2024-07-13:
I'm also having this issue on Prusaslicer 2.8.0 on 64-bit Windows 10.
I'm unable to type "-" or "=" from the numeric row in any non-dialogue-box text field I can find in the slicer. All other keys work, including"_", "+", and "-" on the numpad.

I can't type -/= in print settings fields, notes fields, gcode fields, etc. But if I click to save a preset, a dialogue box pops up, and I can type them there.

Pasting the characters does work in all fields.

Murder-exe, 2024-08-03:
Win11 64 bit:
Won't allow me to enter symbol "-" when using English Language with keyboard Czech (QUERTY), but it works when I switch to English language with keyboard US. Both using in-build laptop keyboard.

karacaenes, 2024-08-19:
win 11, same issue but can't enter "4" anywhere. (i.e. any of the print settings number boxes). Very annoying while using 0.4 nozzle. I can copy a 4 and paste it to places i can't enter with keyboard though. (also, i do not have a numpad on my keyboard, standard number row "4" and "-" do not work.

DaGwhite, 2024-09-09:
It's not just that it's not accepting negative input from the keyboard, it appears to ignore nagative values in existing projects when you load then and try to print.

I use small amounts of negative Deretraction Extra Length ("DEL") to clean up seams, and have used this successfully for a couple years. Since installing 2.8.0, I have been fighting with bad seams reprinting parts that used to be fine. I just discovered that if you don't load an old file with a negative DEL, you cannot type one in.   

That's annoying enough, but I have also verified that 2.8.0 ignores the negative value when you load an existing file containing a negative DEL.  I re-installed 2.7.4, and the same project prints just fine, where the seams are really bad in 2.8.0.

neophyl, 2024-09-09:
Just commented on your forum post but I think your issue is this one https://github.com/prusa3d/PrusaSlicer/issues/11914

TerPuc, 2024-09-10:
Hi @DaGwhite, may I kindly ask you to provide the 3mf file? Thanks!

DaGwhite, 2024-09-10:
I included it with my original post, but that got closed as a duplicate.  I've attached a copy here.  Thanks for investigating this.

Further tests have shown that it is not ignoring the "-" sign when I import my old project.  The ugly seams I get in 2.8.0 must be from something else that was changed from 2.7.4.
[Neg DEL Bug.zip](https://github.com/user-attachments/files/16944298/Neg.DEL.Bug.zip)

lukasmatena, 2024-09-12:
Fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases). Closing.

closed 2024-09-12 by lukasmatena