prusa3d/PrusaSlicer #13045: Binary gcode with no option to not do that

gitcnd, 2024-07-08:
### Description of the bug

Tells me I need some specific firmware to use the output it's going to make.
does not give me any way to tell it not to do that in the popup that shows

This update is terrible - changes all the screen headlines making it extra hard to find anything, removes the "about" option, etc etc etc.

Why on EARTH did so much stuff have to be changed for no reason???

### Project file & How to reproduce

export gcode for 1st time in 2.8

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Build: PrusaSlicer-2.8.0+win64-202406270929

### Operating system

Win 10

### Printer model

ender, but using prusa mini profile

u89djt, 2024-07-08:
Fellow user here looking to save you some stress: the search box top right of the window works well.
[image]
Having been taken aback by the changes myself initially, I'm happy to report that it does allow for smoother running once the basics have been rediscovered. Some reasons are described on the releases page, and some of them become evident with use, I think.

gitcnd, 2024-07-08:
Yeah, nah. whacking that hidden in the options, instead of giving it to me as part of the warning dialogue is just being mean to everyone.

Not to mention - making it a default when only specific firmwares understand it?  Bad idea.  Double-not-to-mention: who even cares?  There was nothing wrong with the ASCII version in the first place, which also included all the version and metadata and settings info in readable text format - now all of that is lost, **for no sane reason**.  If someone wants to save $0.000001 on their SD card storage, that's their obscure problem: shipping a default mass of regressions and incompatibility "features" is just not sensible.

2 more bugs - I cancelled when it said that to me... but it saved anyhow.

I installed the older working version without all the crazy UI regression (phew!) - but even though it *said* my settings had been preserved during the first update, the reverted version was not able to use them (seems stupid - using 2.8 to save 2.7.2 settings in a way that nothing else except 2.8 can read back?  2.7.2 should have saved them for me, when I clicked the "update" button...)

u89djt, 2024-07-08:
You seem to be very sure. Good luck.

closed 2026-09-29 by prusaslicer-bot[bot]