prusa3d/PrusaSlicer #13046: Prusa Connect in Prusa Slicer not working

daren-ouz, 2024-07-08:
### Description of the bug

When I log in to Prusa Connect in Prusa slicer, error is printed in terminal and empty only screen is shown when accessing Prusa Connect in menu bar.

Log in is successful, account name is shown with ability to log out. However, in the terminal from which I start slicer error message appears:

[2024-07-08 11:33:46.406171] [0x00007f315064bd40] [error]   Failed communication with Prusa Account: 0xb7be7f8 action failed (400): {
  "message": "key parameter is invalid"
}

(prusa-slicer:346017): GLib-GObject-WARNING **: 11:33:46.546: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:346017): Gtk-CRITICAL **: 11:33:46.546: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

and Prusa Connect shows only empty screen.

However, under selected printer appears status field (1 printer ready 0 offline 0 printing), so connection to PC somehow works...

### Project file & How to reproduce

[image]
[image]

PrusaSlicer
Version:   2.8.0+linux-x64-GTK3
Build:     PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 6.1.0-22-amd64 x86_64
Total RAM size [MB]: 33,419MB
OpenGL installation
GL version:   4.6.0 (4.6.0 NVIDIA 535.183.01)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 2070/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
[... 362 lines cut ...]
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
</details>

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+linux-x64-GTK3

### Operating system

Debian 12.6

### Printer model

MK4, XL, MINI

DannyH3103, 2024-07-08:
Sign in to [Prusa Connect](https://connect.prusa3d.com) on the web browser, then try to open the PrusaSlicer tab again.

daren-ouz, 2024-07-09:
I'm continuously signed in web browser and I can see all my printers... By logging out in PS a loose Prusa Connect menu button and printer status (as expected), by re-logging in I get menu button (with empty page, as posted above in print screen) and printer status (again, as posted above). No change here ;/

I have tried to log out from PC in browser, test it in PS, log in in browser, test again - no change here...

DannyH3103, 2024-07-09:
It's unfortunate that signing in didn't help. There's another issue for a similar problem that has much more attention. It is on MacOS, but a PrusaSlicer contributor said they will be working on fixing the bug. Maybe you could share your experience there too.
[Here's the issue.](https://github.com/prusa3d/PrusaSlicer/issues/12951)

daren-ouz, 2024-07-10:
OK, I'll wait for next subversion upgrade of PS and test it. If some optional questions will arise, I'll try to answer/test it. Thanks for your effort...

kocikdav, 2024-09-23:
Hello, this seems to be same problem as in https://github.com/prusa3d/PrusaSlicer/issues/12969. Would you please confirm, that your issue is solved by either running 
`$ WEBKIT_DISABLE_DMABUF_RENDERER=1 prusa-slicer`
or
`$ WEBKIT_DISABLE_COMPOSITING_MODE=1 prusa-slicer`
or both? We will than set this in code.

daren-ouz, 2024-09-24:
Hi, I've tried all 4 combinations of variable settings with 2.8.0 and 2.8.1 versions of slicer with no difference. All times I'm signed to my connect account but see only blank page on Prusa connect tab ;/

daren-ouz, 2024-10-24:
Hi all, I have just noticed, that Prusa Connect tab in slicer started to work suddenly. I use the same version of slicer, so I do not know the reason of the change (I use automatic upgrades of OS, so some package could be upgraded, or something on the side od Prusa Connect web pages changed, I do not know).

Thanks for all sugestions...