prusa3d/PrusaSlicer #13048: First layer in vase mode uses travels

cimbalek, 2024-07-08:
### Description of the bug

When printing in vase mode (3DLabPrint plane), 1st layer is not printed in vase mode despite vase mode and 0 bottom layers are set. It then uses unnecessary travels and causes LWPLA adhesion issues.

### Project file & How to reproduce

[vase-travel-reproducer.3mf.zip](https://github.com/user-attachments/files/16134761/vase-travel-reproducer.3mf.zip)
1.) slice
2.) see travel moves in 1st layer

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Linux (fedora)

### Printer model

Ender 3 pro (unrelated)

kubispe1, 2024-07-09:
I am afraid that the Slicing mode Even-odd does not work with the Arachne generator. Search for "perimeter generator" in the PS Search Bar and switch to the Classic Perimeter generator. It worked for me.
- Because I am not an expert in this specific slicing approach, I checked the website and found a note about this here: [3DLabPrint FAQ](https://3dlabprint.com/faq/prusaslicer/). Unfortunately, we have no information about possible problems with the Arachne generator on our help page: [Prusa Knowledge Base](https://help.prusa3d.com/article/slicing-mode-3dlabprint-airplanes_301501).
- Additionally, I am not sure if it is a necessary to combine Even-odd mode and Spiral vase mode, but I understand what you are trying to reach. 
- More or less, I think that the first layer should have a consistent height and not be in spiral mode because of adhesion issues.
- Just an extra note: On the 3DLabPrint website, they recommend using a modifier to put two perimeters for the first layer. This improves adhesion even more. Check out the video here: [YouTube Video](https://youtu.be/Nwm5eX0_pgU).
So, I am closing this issue. If you find any problems, let me know.

closed 2024-07-09 by kubispe1

cimbalek, 2024-07-17:
Thanks for your answer.
Yeah, I might try to turn off arachne perimeter generator. But thatþs all that is applicable from your suggestions:

- Print settings are crafted by 3DLabPrint themselves, because they no longer ship STL files, just 3MF files. So I guess there is a reason for this combination, given even-odd slicing mode was introduced as an aswer for their request.
- Model was developed to be printed as a spiral vase and adhesion between layers is not an issue for LWPLA filament. Issue is traveling because of oozing, that you cannot mitigate with LWPLA. No travels = no oozing. But first layer uses travels because of some reason while it shouldn't and has no reason to do so
- Video is 2 years old and doesn't apply for LWPLA, just for PLA models. Spiral vase models were introduced a year ago so the video cannot contain information for this matter and thus I consider it unrelevant. It's applicable for older model printed from PLA / PolyAir

cimbalek, 2024-08-31:
Can you please reopen this issue? I donþt get why you closed it if you are "not an expert in this specific slicing approach".