prusa3d/PrusaSlicer #13049: prusaslicer closes when selecting finish in configuration wizard

white-stormy, 2024-07-08:
### Description of the bug

when i enter the configuration wizard it will allow me to select and unselect things but when i click on finish the application closes. it does not retain any changes i made.

### Project file & How to reproduce

enter configuration wizard
do anything or nothing, same results
click finish
[project.zip](https://github.com/user-attachments/files/16135266/project.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

windows 10 pro

### Printer model

mk4

kubispe1, 2024-07-09:
@white-stormy What do you mean by  "application closes" ? You mean whole PS Crash. 
If you may record a video, it would be good starting point.

white-stormy, 2024-07-09:
Prusaslicer will crash without throwing an error, so it would act as if you
closed prusaslicer. It would do the same thing in the beta and rc.

kubispe1, 2024-07-10:
I am not able to reproduce it. Please try the following steps:

1) Go to Menu > Help > Show Configuration Folder. This will open the PrusaSlicer folder.
2) Copy this folder to back up your profiles and settings.
3) Delete the PrusaSlicer folder.
4) Close the PrusaSlicer application.
5) Relaunch the Slicer.
6) Try to reproduce the issue.
I hope than everything will be fine.

white-stormy, 2024-07-13:
that fixed it. thank you for the help

closed 2024-07-13 by white-stormy