prusa3d/PrusaSlicer #13050: Prusaslicer 2.8.0 Bizarre issue - Preview window covered in triangles and travel lines

cfisch06, 2024-07-09:
### Description of the bug

When I slice any STL, the preview of the object becomes an intelligible mass of lines criss-crossing the screen. The estimated time seems reasonable, but I can't see the object and I don't want to print for fear of what it could do to the printer. I am using a 2011 MacBook Pro (used Opencore patcher to update computer to Monterey), Prusaslicer 2.8.0, MK3s default profile. This issue occurs with any STL, even ones which worked perfectly well before I updated. 

[image]
[image]

### Project file & How to reproduce

[Pill Case_MK2_Final.3mf.zip](https://github.com/user-attachments/files/16136557/Pill.Case_MK2_Final.3mf.zip)

Any time any stl is sliced this issue appears in the preview page. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+MacOS-x64

### Operating system

macOS Monterey version 12.7.4

### Printer model

Prusa MK3s

kubispe1, 2024-07-09:
I have already come across your issue on Reddit (probably it is yours): [Bizarre PrusaSlicer 2.8.0 issue](https://www.reddit.com/r/prusa3d/comments/1dyqvi0/bizarre_prusaslicer_280_issue/).

The attached STL seems to be different from what you show in the pictures. Your attached model has open edges and includes errors. Never mind, even with the broken STL, I can slice it without any visible problems in 2.8.0 on Windows. I am quite skeptical that this issue could be OS-specific.
Let's try this one:
[Pill Case_MK2_Final_repaired_STl.zip](https://github.com/user-attachments/files/16144125/Pill.Case_MK2_Final_repaired_STl.zip)

cfisch06, 2024-07-09:
> I have already come across your issue on Reddit (probably it is yours): [Bizarre PrusaSlicer 2.8.0 issue](https://www.reddit.com/r/prusa3d/comments/1dyqvi0/bizarre_prusaslicer_280_issue/).
>
> The [...]

Hi kubispe1, I downloaded the file you repaired, however I still get the same issue (see screenshot with result). Note that this issue appears for any STL I've attempted, even some that worked fine a few days ago before I updated to 2.8.0
[image]

kubispe1, 2024-07-09:
Now I am assuming that it is a bizarre issue—some kind of LSD mode viewing. I will forward it. Let's see if we can find a solution. As far as I know, you are the first user with this kind of issue.

cfisch06, 2024-07-09:
Thank you for forwarding it! I’m using an older MacBook Pro 2011 so perhaps it has to do with the older graphics card or gpu/cpu. When I open a previously-successful Gcode in the gcode viewer I get the same issue

DerKameramann, 2024-07-10:
Hi!
cause i just postet my troubles with an old Mac i tried your repaired 3mf and it doesent reproduce your problems here - but i could reproduce my issue. So i agree its a grafics card driver thing.

truckershitch, 2024-07-25:
I am experiencing this issue with 2.8.0 and Fedora 40, rpmFusion nVidia main drivers for a 1060 card.
[image]

It happens with every initial change to the preview window when the program launches.  If I change the filament type, the screen appears as normal.

phoenixphreek, 2024-08-04:
I am having this same issue, also on a 2011 mac. Was there a fix? Should I use an older version of prusa slicer?
[image]
[image]

truckershitch, 2024-08-11:
> I am experiencing this issue with 2.8.0 and Fedora 40, rpmFusion nVidia main drivers for a 1060 card. [...]

Replying to myself here:  I was running Wayland instead of Xorg.  I was also having other video issues.   When I switched back to using Xorg, the issues with PrusaSlicer went away.

cfisch06, 2024-09-12:
> Hi! cause i just postet my troubles with an old Mac i tried your repaired 3mf and it doesent reproduce your problems here - but i could reproduce my issue. So i agree its a grafics card driver thing.

The update Prusaslicer 2.8.1 solves this issue! Now works perfectly on my older Mac.

closed 2024-09-12 by cfisch06