prusa3d/PrusaSlicer #13051: turn off purge tower button for single color prints

Peewee0512, 2024-07-09:
hello i think it would help if there was a way to turn off the purge tower from the pre sliced bed because if I'm doing single color prints it takes a lot of room and then is just gone

kubispe1, 2024-07-09:
I accept your request, but it would be nice if PS could automatically detect the used extruders on the build plate and, accordingly, switch off the tower for single-color prints.

kassi, 2024-07-09:
I started to use my MMU configuration for single color prints recently to be able to start the print right away without having to manually select the filament at the printer.
Interestingly there was no purge tower present at all for any for my single color prints. Looks like PS is already able to do that? The preview doesn't show a tower either, but I'm currently not sure if it's visible in multi color prints.

kubispe1, 2024-07-10:
What Printer prest Do you use @kassi? MK3S+MMU3 Single...there it is fine
**XL 5T**
[image]
**MK4MMU3**
[image]

kassi, 2024-07-11:
I'm using MK4+MMU3. If it's single color, no tower visible. As soon as I change one color it appears. If I re-choose to have only one color, it disappears.
[image]
[image]

techis, 2024-08-17:
If it helps, the MMU3 for the Mk3S+ had the option to use MMU Single filament. Would be nice if that was an option for the Mk4.

Jan-Soustruznik, 2025-08-21:
cloased as duplicate of #7813

closed 2025-08-21 by Jan-Soustruznik