prusa3d/PrusaSlicer #13052: add plates like Bambu studio. like to have 3d beds ready to print

Peewee0512, 2024-07-09:
**hello so I know the title may be confusing it's kind of hard to explain but like the Bambu slicer we can have multi beds with objects and can " print off each bed" I think i9t would help if I have parts I need printed and I could go ahead and slice them and add them to like a **queue** but like be able to see all beds at one because something I have files that have 20 parts and that can't all fit on one bed so it would be nice to have like 3 beds with the parts spaced  out and I can pick which bed I would like to print** 
[image]
[image]

u89djt, 2024-07-09:
(Fellow user always looking for workarounds) You can achieve a chunk of that function now UNLESS you use auto-arrange. Here's an animated gif: 
[image]
To select a bed full of objects, hold the shift key first, and top-down view (1) is safest I guess. Ctrl A to select everything to shift a different bed's-worth onto the bed. The wipe tower seems to confuse the selection process - or I'm confused - so there may be a bit more thinking to do.
It would be good to be able to tinker with auto arrange on multiple beds. A key reason for being able to do it all in one instance of the slicer is to be able to switch objects around between beds without friction. I guess if auto arrange could be told to only operate within a single bed, stuff could be dragged into a heap in the middle then arranged. I imagine this is already on the list somewhere :)

kubispe1, 2024-07-09:
Thanks. There already exists a thread: #10704. I am closing this issue. See you.

closed 2024-07-09 by kubispe1