prusa3d/PrusaSlicer #13053: Could not find a package configuration file provided by "boost_nowide"

Haidiye01, 2024-07-09:
### Description of the bug

>-- CMAKE_PREFIX_PATH: E:\work\PrusaSlicer-deps\usr\local (from cache or command line)
9>/Include//winrt/windows.graphics.printing3d.h was not found
9>STL fixing by WinSDK will not be compiled
9>Building without Win10 STL fixing service support
9>CMake Error at E:/work/PrusaSlicer-deps/usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake:141 (find_package):
9>  Could not find a package configuration file provided by "boost_nowide"
9>  (requested version 1.78.0) with any of the following names:
9>
9>    boost_nowideConfig.cmake
9>    boost_nowide-config.cmake
9>
9>  Add the installation prefix of "boost_nowide" to CMAKE_PREFIX_PATH or set
9>  "boost_nowide_DIR" to a directory containing one of the above files.  If
9>  "boost_nowide" provides a separate development package or SDK, be sure it
9>  has been installed.
9>Call Stack (most recent call first):
9>  E:/work/PrusaSlicer-deps/usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake:262 (boost_find_component)
9>  CMakeLists.txt:380 (find_package)

### Project file & How to reproduce

>-- CMAKE_PREFIX_PATH: E:\work\PrusaSlicer-deps\usr\local (from cache or command line)
9>/Include//winrt/windows.graphics.printing3d.h was not found
9>STL fixing by WinSDK will not be compiled
9>Building without Win10 STL fixing service support
9>CMake Error at E:/work/PrusaSlicer-deps/usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake:141 (find_package):
9>  Could not find a package configuration file provided by "boost_nowide"
9>  (requested version 1.78.0) with any of the following names:
9>
9>    boost_nowideConfig.cmake
9>    boost_nowide-config.cmake
9>
9>  Add the installation prefix of "boost_nowide" to CMAKE_PREFIX_PATH or set
9>  "boost_nowide_DIR" to a directory containing one of the above files.  If
9>  "boost_nowide" provides a separate development package or SDK, be sure it
9>  has been installed.
9>Call Stack (most recent call first):
9>  E:/work/PrusaSlicer-deps/usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake:262 (boost_find_component)
9>  CMakeLists.txt:380 (find_package)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8

### Operating system

win10

### Printer model

no

lukasmatena, 2024-07-10:
The deps folder should contain boost 1.83. Why do you have 1.78 in there? Did you rebuild the dependencies?

Haidiye01, 2024-07-10:
ok,i have received your letter ,thank you!

lukasmatena, 2024-07-10:
Closing, this will almost certainly disappear after deps are rebuilt.

closed 2024-07-10 by lukasmatena