prusa3d/PrusaSlicer #13054: Not possible to enter a "1" in any parameter field.

rfeier, 2024-07-09:
### Description of the bug

When i have to enter a "1" in a parameter field, noting happens. all other digits are working. I have to copy-paste from outside the program.

### Project file & How to reproduce

this happens with all files. and a restart of the program doesnt solve the problem. I uninstalled and reinstalled the program no difference. Even on another PC i have the same problem.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 11

### Printer model

MK4

neophyl, 2024-07-09:
https://github.com/prusa3d/PrusaSlicer/issues/12961
There are also several more **existing** issues regarding some characters not being possible to enter in various methods.  A nice list is in https://github.com/prusa3d/PrusaSlicer/issues/13043

lukasmatena, 2024-07-10:
Closing as a duplicate.

closed 2024-07-10 by lukasmatena