prusa3d/PrusaSlicer #13056: Smaller layer height with input shaping with Prusa XL

Peewee0512, 2024-07-09:
**just I would like a smaller layer Height with input shaping so I can get that extra fine detail fast... maybe 0.08 or smaller thanks**

u89djt, 2024-07-10:
Hi there, (fellow user thinking) that's already something you can do:
[image]
I guess there's extra motivation to perform extrusion multiplier calibration for your specific filament to make it as accurate as possible.
https://help.prusa3d.com/article/extrusion-multiplier-calibration_2257

Peewee0512, 2024-07-10:
yeah but like a saved preset.

u89djt, 2024-07-10:
You can do that too :) 
[image]
For inbuilt presets, maybe they'd be asking themselves where does this end? Maybe 0.1 is the smallest layer height they feel secure offering for users who don't prioritize filament quality and maintenance/calibration.

kubispe1, 2024-07-10:
Hi, thanks for the clarification on @Peewee0512's question. It does not seem like a bug or issue to me. I am closing it.
@u89djt, can you send me an email at petr.kubis@prusa3d.cz just for discussion?

closed 2024-07-10 by kubispe1

u89djt, 2024-07-10:
@kubispe1 tis done!